jisotalo/ads-client #10: Add support for multi-dimensional arrays

jisotalo, 2020-04-16:
At the moment multi-dimensional arrays are not handled correctly. Datatype information includes correctly parsed multi-dimensional array data, but it's not parsed to Javascript object correctly.

Example:
`ARRAY [0..9,1..5] OF STRING(80) 
`
Type:
```
{ name: '',
  type: 'STRING(80)',
  size: 81,
  offset: 0,
  adsDataType: 30,
  adsDataTypeStr: 'ADST_STRING',
  comment: '',
  attributes: [],
  arrayData:
   [ { startIndex: 0, length: 10 }, { startIndex: 1, length: 5 } ],

```

Result as javascript:
`[ '', '', '', '', '', '', '', '', '', '' ]`

jisotalo, 2020-05-24:
Coming up in 1.4.0

jisotalo, 2020-05-28:
Released in 1.4.0

closed 2020-05-28 by jisotalo

Machy76, 2020-06-08:
Hi,

thank you for ads-client. Good work.

But now I trying reading multidimensional array of LREAL inside struct and there is some issue with data type. See attached image.

Machy

[image]

jisotalo, 2020-06-08:
Hi @Machy76 and thanks for the feedback!

So the array is inside a struct. Is there a `{attribute 'pack_mode' := '1'}` above in that struct definition and other structs that are used inside it?

If there is and it's not working, could you provide a simple example struct that is not working? I will definetely look into it!

Machy76, 2020-06-08:
I try it on simple struct:

> {attribute 'pack_mode' := '1'}
TYPE ST_Dosing_WorkInSave_1 :
STRUCT
	bHSOnOff:								ARRAY[1..2, 1..6] OF BOOL;						(*zapnuti rady*)
	fHSDoseSet:								ARRAY[1..2, 1..6] OF LREAL;					(*nastavena velikost davky*)
END_STRUCT
END_TYPE

And use this in this variable "Dosing_1":

> VAR_GLOBAL
END_VAR
VAR_GLOBAL PERSISTENT
	WorkInSave:									ST_Dosing_WorkInSave_1;
END_VAR

I attached some images with state in TwinCat and response with Postman. You can see that a change in bHSOnOff will cause a change in fHSDoseSet.
[image]
[image]
[image]

jisotalo, 2020-06-08:
Thanks! I will check this today evening after getting from work.

If possible, could you please try in the meantime if array of REAL values works or has the same problem? It might be a problem with LREAL values only or otherwise it's a problem with arrays.

Machy76, 2020-06-08:
Thank for fast feedback!

Array of REAL has same problem as LREAL...

jisotalo, 2020-06-08:
Ok, thanks. I will get back to you later today!

jisotalo, 2020-06-08:
Hi @Machy76! I just released a new version 1.5.1 where this should be fixed.

I had forgotten to update one calculation that was required for multi-dimensional arrays... It caused reading multi-dimensional arrays to fail when they were inside a struct.

For my tests it works OK. How about you?

Thanks for reporting this!

Machy76, 2020-06-09:
I tested reading multidimensional arrays inside bigger structure and with version 1.5.1 it works OK.

Thank You very much!

jisotalo, 2020-06-09:
Great!