jisotalo/ads-client #17: MultiRead and MultiWrite

aronneGrinton, 2020-04-25:
Hi,
very nice work!!

What about to add support for multiRead / multiWrite ecc..?

For example for multiRead something like

`    /**
   * Reads (raw byte) data from PLC by given index group, index offset and size
   * 
   * All required parameters can be read for example with getSymbolInfo() method, **see also readRawBySymbol()**
   * 
   * @param {Arry} variablesArray - Array of objects {indexGroup : (number), indexoffset : (number), byteSize (number)}
   * - If resolved, reading was successful and data is returned (Buffer)
   * - If rejected, reading failed and error info is returned (object)
   */
  readRawMulti(variablesArray) {
    return new Promise(async (resolve, reject) => {

      if (variablesArray == null) {
        return reject(new ClientException('readRaw()', `Some of parameters (variablesArray) are not assigned`))
      }

      debug(`readRawMulti(): Reading data from ${JSON.stringify({ variablesArray })}`)
      
      let dataSize = 0;
      for(let i = 0; i < variablesArray.length; i++){
        dataSize = dataSize + variablesArray[i].byteSize
      }

      //Allocating bytes for request
      const data = Buffer.alloc(16 + variablesArray.length*12)
      let pos = 0

      //0..3 IndexGroup: reserved for SumCommandRead
      data.writeUInt32LE(ADS.ADS_RESERVED_INDEX_GROUPS.SumCommandRead, pos)
      pos += 4
      
      //4..7 IndexOffset: number of SumCommands
      data.writeUInt32LE(variablesArray.length, pos)
      pos += 4
      
      //8..11 Read length: Length of data in bytes, which should be read. RequestedDataSize + Length of requested data + 4 byte errorcode per sum_command;
      data.writeUInt32LE(dataSize + 4 * variablesArray.length, pos)
      pos += 4

      //12..15 Write Length: Length of data in bytes, which should be written. 12*sum_command
      data.writeUInt32LE(12 * variablesArray.length, pos)
      pos += 4

      //16... Data: Data which are written in the ADS device.
      for(let i = 0; i < variablesArray.length; i++){
        
        data.writeUInt32LE(variablesArray[i].indexGroup, pos)
        pos += 4

        data.writeUInt32LE(variablesArray[i].indexOffset, pos)
        pos += 4

        data.writeUInt32LE(variablesArray[i].byteSize, pos)
        pos += 4
      }

      
      _sendAdsCommand.call(this, ADS.ADS_COMMAND.ReadWrite, data)
        .then((res) => {
          debug(`readRaw(): Data read - ${res.ads.data.byteLength} bytes from ${JSON.stringify({ variablesArray })}`)
          
          resolve(res.ads.data)

        })
        .catch((res) => {
          debug(`readRaw(): Reading data from ${JSON.stringify({ variablesArray })} failed: %o`, res)
          reject(new ClientException('readRaw()', `Reading data failed`, res))
        })
    })
  }

`

And for example testing with 2 INT:

`client.readRawMulti([{indexGroup: 16448,indexOffset: 387930, byteSize: 2}, {indexGroup: 16448,indexOffset: 400092, byteSize: 2}])
    .then(res => {

      var myValues = [];   
      var pos = 0;  

      // Check ads errors
      var checkErrors = [];

      for (var i = 0; i<2; i++) {
        checkErrors.push(res.readUInt32LE(pos))
        pos += 4
      }

      // Reading 2 INT
      for (var i = 0; i<2; i++){
        myValues.push(res.readInt16LE(pos))
        pos += 2
      }
      
      console.log('Errors : ' + checkErrors  )
      console.log('Values : ' + myValues)

    })
    .catch(err => {
      console.log('Something failed:', err)
    })`

jisotalo, 2020-04-26:
Thanks for the feedback, hope you have use for this library! And thank you for this feature request.

That's a great idea indeed. I will look into it and and probably add it quite soon. It could also be later implemented to the `readSymbol` and the `writeSymbol` methods to support multiple symbols with less traffic and code. Thanks for providing the base code!

aronneGrinton, 2020-04-26:
Consider to use MultiCommands/Sum_up/Sum_command also to add notification events, or to read/ask for multyple handles ecc...

Raw methods are very useful. Using them you can also ask complex data structures (for example array of structures or multidimensional array) and riceve the data (
though you have to decompose the recived data properly to match the structure).

jisotalo, 2020-05-17:
I'm developing these in the next release 1.3.0. The auto-reconnect took some time so sorry for the delay.

jisotalo, 2020-05-24:
@aronneGrinton 
Released in 1.3.0 today. Creating/deleting multiple variable handles is coming later.

closed 2020-05-24 by jisotalo