jisotalo/ads-client #27: change twincat state

MaximeMylle, 2020-05-26:
Hi,

first of all, sorry that im making an issue for this, i didn't know any other way to contact.
second of all, very impressive work you've done here, im using this npm package and s working great for me.

last of all, just a question because i cant find any documentation on it. I see the Twoncat state can be read, but not written. Is this possible at all?

to be more specific, i would like to make a "software-upgrade" trough a webinterface, but for that i need to set the state from RUN to CONFIG

kind regards

Maxime

MaximeMylle, 2020-05-26:
I see issue 24 already handles this question, sorry to bother, and really awaiting this feature

closed 2020-05-26 by MaximeMylle

jisotalo, 2020-05-26:
Thank you for your feedback! Great to hear that the library is in use!

I will implement the run<->config change command, if it's possible. See #24 .

jisotalo, 2020-05-30:
Released in 1.5.0. See readme for details:

https://github.com/jisotalo/ads-client#starting-and-stopping-the-plc
https://github.com/jisotalo/ads-client#starting-and-stopping-the-twincat-system