jisotalo/ads-client #36: readAndCacheDataTypes() and getDataType() inconsistency

DmitrijTrifanov, 2020-06-21:
Good day.

Noticed something interesting - call to .getDataType(...) produces a structure, which has two fields called "adsDataType" and "adsDataTypeStr". Now from what I can see, a call to .readAndCacheDataTypes()  produces a map, with children structures, identical in their content to the call to .getDataType(), with the exception that "adsDataType" and "adsDataTypeStr" are keyed as "dataType" and "dataTypeStr". Outside of that, both results, in terms of their schema, look identical. 

Wondering if that difference is intentional or not? The reason is, if parsing the type information, two difference parser functions need to be made to account for the difference in the names, which doesn't seem necessary, unless of course, I am missing the reason for the different names.

jisotalo, 2020-06-21:
Hi @DmitrijTrifanov!

Thank you for your feedback. You are absolutely right, they are the same. This is just a mistake during development. It is the data type used in ADS as a enumeration and then as string.

I will change the `dataType` and `dataTypeStr` to `adsDataType` and `adsDataTypeStr` in the next release, as they are more descriptive. The symbol PLC data type is available in `type` variable and it will be so in the future too.

DmitrijTrifanov, 2020-06-21:
Great! That will save me one function call, but having consistency between the two schemas, should make it easier to keep track of things.

jisotalo, 2020-06-22:
Fixed in version 1.6.0

closed 2020-06-22 by jisotalo