jisotalo/ads-client #39: Read Attributes of Data readAndCacheSymbols() Method

DmitrijTrifanov, 2020-06-23:
Good day

I have been trying to work out if this is a limitation of ADS when it comes to Symbols located directly in Program file types, or is the data simply not fetched. When receiving data with the `.readAndCacheSymbols()` method, the data contains the 'Attributes' entry in flagsStr, but the attributes field itself seems to be missing. By comparison subItems which are gathered by get data types in structured value, has all the attributes in { key : string, value : string } format.

Is that data unavailable through ADS by any chance?

jisotalo, 2020-06-24:
Hi @DmitrijTrifanov!

Thanks for your valuable comments!

I checked into this and yes, it's actually missing from the code. When using `readAndCacheSymbols()` or `getSymbolInfo()`, you will get symbol information. The information received from ADS also contains array info, attributes and some other data, but they are not parsed from byte buffer at the moment.

This is easy to implement, it is missing because the idea was to impelement it later and then I forgot (as it's not needed by ads-client itself). I will release it later today!

Even the comment in the code says:

```
//The rest of the data would contain ArrayInfo, TypeGuid, Attributes and reserved bytes
//Skipping for now
```

DmitrijTrifanov, 2020-06-24:
Thanks for such a quick response! Glad to hear it’s not a limitation of ADS and it’s not a lot of effort to implement.  Looking forward to the update

jisotalo, 2020-06-24:
Hi @DmitrijTrifanov! 
I added this to 1.7.1. Could you confirm it fixes this issue?

Thanks!

DmitrijTrifanov, 2020-06-24:
Yep. Works great now

jisotalo, 2020-06-24:
Great!

closed 2020-06-24 by jisotalo