jisotalo/ads-client #46: Reading symbol info failed [TC3]

dotKokott, 2020-09-28:
Hey, first of all for this great looking project!

I have a problem getting the example to run. While I can connect to my local target (after enabling the TPC loopback) I cannot seem to read any variables. I always get:
`Something failed: ClientException: Reading symbol PLC1.GLOBALS.FAKE_SUB_SYSTEM_COUNTER_MAX failed: Reading symbol info failed`

The symbols are mapped when I check them through the HMI Configuration Window:
[image]

Is there another step necessary to expose them somehow to the ADS interface?

dotKokott, 2020-09-28:
After running client.readAndCacheSymbols() I realized that my symbols were there but without the namespace 'PLC1' in my case.

So reading the symbol 'GLOBALS.FAKE_SUB_SYSTEM_COUNTER_MAX' works! A bit confusing because according to the configuration it is mapped to 'PLC1.GLOBALS.FAKE_SUB_SYSTEM_COUNTER_MAX' but I am happy it works!

Thank you again!

closed 2020-09-28 by dotKokott

jisotalo, 2020-09-28:
Hey! Great that you solved it and thanks for the feedback :)

I was just going to reply that try without the PLC1. The ADS communication namespace seems to always start without any device/project names, that can be seens sometimes in the Visual Studio. Same for example with C# library.