jisotalo/ads-client #48: Fails to write on large number of batch writes.

jhenson29, 2020-12-03:
## Issue

I've been testing how many tags I can read/write at the same time by mapping over an array of tags to read or write inside of a `Promise.all`. It varies on how many it can do at a time. Sometimes it's 400. Sometimes it's 800. I've found that if I batch them in smaller quantities, then it's not an issue. 200 tags at a time appears to work consistently at the moment.

To test, I setup an array of 1000 DINTs and tried to write them as individual tags at the same time.

The error message thrown from `readSymbol` is `Tag write failed Writing symbol Main.testRead[778] failed: Writing the data failed`. (Specific to the particular tag)

## General Information
**Target**
- CX-5130
- TwinCAT 3.1 4024.7

**Local Dev**
- Windows 10
- TwinCAT 3.1 4024.10
- NodeJS v12.16.1
- ads-client v1.10.1

## Examples

I have a TypeScript wrapper around the ads client.

`writeTag` method:

```typescript
    public writeTag = async ({name, value}: ITag) : Promise<boolean> => {
        try {
            await this._client.writeSymbol(name, value);
            return true;
        }
        catch {
            return false;
        }
    };
```

`writeTagGroup` method where `ITagGroup` is an array of `ITag`: (doesn't write all values)
```typescript
    public writeTagGroup = async (tagGroup: ITagGroup) : Promise<boolean> => {
        try{
            const result = await Promise.all(tagGroup.map(this.writeTag));
            return result.every(isTrue);
        }
        catch{
            return false;
        }
    }
```

batched `writeTagGroup` method: (appears to work so far)

```typescript
    public writeTagGroup = async (tagGroup: ITagGroup) : Promise<boolean> => {
        try{
            const writeGroup = [...tagGroup];
            const result : boolean[] = [];
            while(writeGroup.length){
                result.concat(await Promise.all(writeGroup.splice(0,200).map(this.writeTag)));
            }
            return result.every(isTrue);
        }
        catch{
            return false;
        
```

jisotalo, 2020-12-04:
Hi Jeremy!

I'm quite sure the reason for this is that the TCP window/buffer/whatsoever of the operating system gets full. But of course, there could be other alternative reasons. Please `console.log` the error that is thrown from the writeSymbol, as it contains more information like ADS error code (if any).

The writeSymbol reads the symbol info for given variable before writing, if it's not cached already. By calling the method 1000 times with different variable, it first downloads symbol information for each variable [0], [1] and so on. So in addition to the data writing, it reads information from the PLC.

I tried to write 1000 DINTs on my laptop to the local TwinCAT, it worked fine. When I increased it to 10000, the first ~1100 worked and it failed then. The error was timeout. When using Wireshark to spy the communication, I could see that the TCP window was full. Basically that means that there was 10000 incoming symbol information ADS responses (160 bytes each) in a very short time. The operating system couldn't handle that. And I also debugged the incoming data in the Node.js, and noticed that I received only like 200k bytes. When I added a 1ms break between requests, it worked fine.

If you really want to do this one by one, instead of struct or array, the only thing I can suggest is to use `writeRawMulti` method. It uses a single ADS request to write all 10k data. Another possibility is to have small, like 1 ms, delays between requests.

```
//NOTE: There is a MAIN program @ PLC with variable TestArray : ARRAY[0..9999] OF DINT;

//Reading symbol info for the first array index only
let symbol = await client.getSymbolInfo(`MAIN.TestArray[0]`)
let multiWrite = []

//Creating a multi write object for each index (increasing offset manually)
for (let i = 0; i < 10000; i++) {
  multiWrite.push({
    indexGroup: symbol.indexGroup,
    indexOffset: symbol.indexOffset + i * 4, //4 = DINT is 4 bytes, you could also use client.getDataType('DINT')
    data: await client.convertToRaw(i * 100, 'DINT') //Using just i*100 as value to write
  })
}

//Now we have an array of 10k write requests -> write it
await client.writeRawMulti(multiWrite)
```

jhenson29, 2020-12-04:
Thank you for your reply. 

The error message I listed at the top was the error.message rejected from the readSymbol promise. But I’ll log the whole error object to see if there is anything else. 

I read the array first, so I assume the symbols were already cached.  But I also see similar problems with reading and writing so some of those reads likely failed and were not actually cached. I would also try reading the group several times in a row and it would sometimes all go thing and sometime fail at different quantities. 

I was also using a remote connection to the PLC, but I had similar issues using TwinCAT on my laptop and connecting to localhost. 

This was just done using an array as a quick, general test for larger numbers of individual tags. In practice, these tags would not be in contiguous memory. The individual tags are gathered from the individual HMI objects and sent to the PLC service for reading / serving. Generally, I don’t need to write large amounts of data. Usually just one tag at a time for button press or numeric entry. But I do read larger quantities of tags for various displays. And the issues I saw were really for both read and write. I just opened the issue under the write operation. 

I’m traveling today but I’ll do more testing tomorrow.

jisotalo, 2020-12-04:
Yeah the error.message just displays that generic message. Under error object you will find more information.

If you read the array using the same, separate variable names like [1], [2] etc, then the symbol information should be cached. If thats so, the write responses filled the TCP window, I guess. In my tests, the library can handle quite a lot data but in this case the node.js socket seems to receive only a small part of the data that should be received.

For fast data reading I would suggest you to try the `subscribe` method, as then only the PLC will send the data and there is 50% less requests.

jhenson29, 2020-12-04:
Thank you for this additional information. I will look at the subscribe method more. I didn’t initially because the driver I use for EIP just uses simple polling as a subscribe method and isn’t any better than just doing group reads. But the EIP driver also implements group reads differently. Group reads send as many tag requests in a packet as will fit. But I think I can implement subscribe in my ADS wrapper without any changes upstream in my PLC service manager.

jhenson29, 2020-12-04:
There are two different types of errors I see from both Read and Write. 2000ms Timeout and Mailbox Full.

**Read Error - 2000ms Timeout**
```
ClientException: Reading symbol Main.testRead[659] failed: Reading symbol info failed
    at C:\Users\jhenson\projects\node-ads-client\node_modules\ads-client\src\ads-client.js:1146:23
    at runNextTicks (internal/process/task_queues.js:62:5)
    at listOnTimeout (internal/timers.js:518:9)
    at processTimers (internal/timers.js:492:7) {
  name: 'ClientException',
  sender: 'readSymbol()',
  adsError: true,
  adsErrorInfo: {
    adsErrorType: 'ADS error',
    adsErrorCode: -1,
    adsErrorStr: 'Timeout - no response in 2000 ms'
  },
  metaData: null,
  errorTrace: [
    'getSymbolInfo(): Reading symbol info for Main.testRead[659] failed',
    '_readSymbolInfo(): Reading symbol information for Main.testRead[659] failed',
    '_sendAdsCommand(): Timeout - no response in 2000 ms'
  ],
  getInnerException: [Function]
}
```

**Read Error - Mailbox Full**
```
ClientException: Reading symbol Main.testRead[199] failed: Reading value failed
    at C:\Users\jhenson\projects\node-ads-client\node_modules\ads-client\src\ads-client.js:1156:23
    at runMicrotasks (<anonymous>)
    at runNextTicks (internal/process/task_queues.js:62:5)
    at processImmediate (internal/timers.js:429:9)
    at process.topLevelDomainCallback (domain.js:137:15) {
  name: 'ClientException',
  sender: 'readSymbol()',
  adsError: true,
  adsErrorInfo: {
    adsErrorType: 'AMS error',
    adsErrorCode: 1282,
    adsErrorStr: 'The mailbox has reached the maximum number of possible messages. The current sent message was rejected'
  },
  metaData: null,
  errorTrace: [
    'readRaw(): Reading data failed',
    '_sendAdsCommand(): Response with AMS error received'
  ],
  get
```

**Write Error - 2000ms Timeout**
```
ClientException: Writing symbol Main.testRead[871] failed: Reading symbol info failed
    at C:\Users\jhenson\projects\node-ads-client\node_modules\ads-client\src\ads-client.js:1227:23
    at runNextTicks (internal/process/task_queues.js:62:5)
    at listOnTimeout (internal/timers.js:518:9)
    at processTimers (internal/timers.js:492:7) {
  name: 'ClientException',
  sender: 'writeSymbol()',
  adsError: true,
  adsErrorInfo: {
    adsErrorType: 'ADS error',
    adsErrorCode: -1,
    adsErrorStr: 'Timeout - no response in 2000 ms'
  },
  metaData: null,
  errorTrace: [
    'getSymbolInfo(): Reading symbol info for Main.testRead[871] failed',
    '_readSymbolInfo(): Reading symbol information for Main.testRead[871] failed',
    '_sendAdsCommand(): Timeout - no response in 2000 ms'
  ],
  getInnerException: [Function]
}
```

**Write Error - Mailbox Full**
```
ClientException: Writing symbol Main.testRead[825] failed: Writing the data failed
    at C:\Users\jhenson\projects\node-ads-client\node_modules\ads-client\src\ads-client.js:1284:23
    at runMicrotasks (<anonymous>)
    at runNextTicks (internal/process/task_queues.js:62:5)
    at processImmediate (internal/timers.js:429:9)
    at process.topLevelDomainCallback (domain.js:137:15) {
  name: 'ClientException',
  sender: 'writeSymbol()',
  adsError: true,
  adsErrorInfo: {
    adsErrorType: 'AMS error',
    adsErrorCode: 1282,
    adsErrorStr: 'The mailbox has reached the maximum number of possible messages. The current sent message was rejected'
  },
  metaData: null,
  errorTrace: [
    'writeRaw(): Writing data failed',
    '_sendAdsCommand(): Response with AMS error received'
  ],
  getInnerException: [Function]
}
```

jisotalo, 2020-12-05:
Thanks! I'm pretty sure that the reason is too many requests in a such small time. On my laptop, I got only the timeout. The mailbox error speaks for itself, so in that case the AMS router got too full of requests or something.

The timeout means that ads-client didin't receive any response in given time (so it's not ADS error, it's our own). So probably the reason is the same on my computer - the client only receives first x bytes until the operating system TCP/IP socket fails (or something like that). In this case it won't help, as there won't be more data, but the timeout delay can be increased if required by giving setting `timeoutDelay` a different value than default 2000 (ms).

closed 2020-12-05 by jhenson29