jisotalo/ads-client #49: Duplicate `Tc3_Plc4` keys in `ADS_RESERVED_PORTS`.

jhenson29, 2020-12-05:
# Issue

Duplicate keys were found for `Tc3_Plc4` in `ADS_RESERVERD_PORTS` in `./src/ads-client-ads.js'.

https://github.com/jisotalo/ads-client/blob/master/src/ads-client-ads.js#L154

```javascript
  //Tc3 PLC RuntimeSystem 4 (Port 854)
  Tc3_Plc4: 854,
  //Tc3 PLC RuntimeSystem 5 (Port 855)
  Tc3_Plc4: 855,
```

# Suggested Fix

Change the second key to `Tc3_Plc5`

jisotalo, 2020-12-06:
Thanks! I'll fix this soon. Also noticed one bug in `WriteControl` and fixing these in the same update.

jhenson29, 2020-12-06:
I found this by starting to convert the project to TypeScript, which I'm mostly doing to familiarize myself with the project.

I found another issue I'll open in a little bit, but I need finish tracing it out first to make sure I'm reporting it correctly. 

Is there any interest in making this TypeScript later?

jisotalo, 2020-12-06:
Fixed in 1.10.2. thanks for reporting the bug!

I haven't had time to learn Typescript yet but it might be possible some day in the future. I'm at the moment working on a ADS server library and wondering if it could be in Typescript, we will see.

closed 2020-12-06 by jisotalo