jisotalo/ads-client #51: Connect to targetAmsNetId on OSX Catalina fails

hansipete, 2021-01-11:
Hello there,

first of all – thank you for that awesome library. 

I'm using it to talk a Beckhoff PLC and it works like a charm. On one of my computers I run into a problem. Even the minimal example would not connect:

```
const ads = require('ads-client')

const client = new ads.Client({
  targetAmsNetId: '5.85.243.162.1.1',
  targetAdsPort: 851,
})

client.connect()
  .then(res => {   
    console.log(`Connected to the ${res.targetAmsNetId}`)
    console.log(`Router assigned us AmsNetId ${res.localAmsNetId} and port ${res.localAdsPort}`)

    return client.disconnect()
  })
  .then(() => {
    console.log('Disconnected')
  })
  .catch(err => {
    console.log('Something failed:', err)
  })
```

Results in this error message:
`Something failed: ClientException: Connection to localhost:48898 failed (socket error -61) (connect ECONNREFUSED 127.0.0.1:48898)`

(OSX Catalina, internal Firewall is turned OFF, no virus scanners etc.)

**I wonder, why this tries to connect to localhost instead of the NetId.** Is this expected behaviour? On the other machine (Windows) it connects just fine, printing:

```
Connected to the 5.85.243.162.1.1
PLC Router assigned us AmsNetId 169.254.105.221.1.1 and port 40334
```

Thanks in advance
Hans

jisotalo, 2021-01-11:
Hi Hans!

Thanks! Great that it has been working so far, glad to hear!

As you OSX doesn't have a TwinCAT router installed, it has no idea where that machine with AmsNetId 5.85.243.162 is (as AmsNetId is not IP address).

As default, the ads-client tries to connect to local router (localhost), which is usually OK when you have TwinCAT installed. However, now you won't have any.

Instead, you need to tell the client to connect to router of another machine, as with Raspberry Pi example. You also need to provide AmsNetId of your own. So follow the https://github.com/jisotalo/ads-client#setup-3---connecting-from-any-nodejs-supported-system-to-the-plc

Change the PLC_IP_ADDRESS to your PLC IP. And the port 48898 needs to be open from firewall.
```
const client = new ads.Client({
  localAmsNetId: '192.168.1.10.1.1',  //Can be anything but needs to be in PLC StaticRoutes.xml file
  localAdsPort: 32750,                //Can be anything that is not used
  targetAmsNetId: '5.85.243.162.1.1', 
  targetAdsPort: 851,
  routerAddress: 'PLC_IP_ADDRESS',     //PLC ip address
  routerTcpPort: 48898                //PLC needs to have this port opened. Test disabling all firewalls if problems
})

```

Please let me know how it's going! I haven't heard any comments from OSX yet.

hansipete, 2021-01-11:
Hi there,

thank you for your explanation! Just figured out that I actually never connected with OSX before... I thought I did, but that was when I was running bootcamped Windows (with TwinCAT installed.)

I'll try to get it running following your guide and keep you updated

Thanks for you help!
Hans

hansipete, 2021-01-11:
Got it working!

It's actually quite easy if using the Beckhoff Web Configuration. `http://<PLC_IP>/config/` (in my case PLC IP = 192.168.99.208) asks for username and password. Then navigate to `config/#TwinCAT&Connectivity` and add my device. (Under the hood this seems to add to TwinCAT\3.1\Target\StaticRoutes.xml)

- AMS Net ID: **192.168.99.106.1.1** (made this one up, using the IP + .1.1)
- Transport Type: **TCP_IP**
- Address: **192.168.99.106**
- Connection Timeout (ms): **60000**
- Flags: **Static, IP Address**

Now I can connect from OSX with the following credentials:
```
{
  localAmsNetId: '192.168.99.106.1.1',  
  localAdsPort: 32750,                
  targetAmsNetId: '5.85.243.162.1.1',
  targetAdsPort: 851,
  routerAddress: '192.168.99.208',  // plc ip   
  routerTcpPort: 48898                
}
```

Works! Thank you so much

jisotalo, 2021-01-11:
Awesome! Thanks for detailed info. I'll link to this issue later from the FAQ if that helps somebody!

I have been thinking about a node.js based AMS router, that would work in any machine (similar to the .NET Core version by Beckhoff). That would be another solution too.

closed 2021-01-11 by jisotalo

Vinhtu, 2022-04-03:
hi can you tell me what platform you are using to connect, i use react-native and it gives error:
TypeError: undefined is not a constructor (evaluating 'new (_$$_REQUIRE(_dependencyMap[13], "net").Socket)()')
Pls help me and thank you very much

Dominic-Liu, 2023-11-24:
> Hi Hans!
>
> Thanks! Great that it has been working so far, glad to hear!
>
> As you OSX doesn't have a TwinCAT router installed, it has no idea where that machine with AmsNetId 5.85.243.162 is (as [...]

>sorry for the trouble, i got the same issue, i don't know where i can perform the code block begins with const client = new ads..., 
could you make some screen shot for me, appreciate your great support.

Dominic-Liu, 2023-11-24:
> Got it working!
>
> It's actually quite easy if using the Beckhoff Web Configuration. `http://<PLC_IP>/config/` (in my case PLC IP = 192.168.99.208) asks for username and password. Then navigate to [...]

Hi Hans, 
glad to find the topic here, could you share me how to use the Beckhoff Web Configuration, if we need to install a certain software?, i have installed the windows in VM, and installed the twincat3 4024.50+ in VM windows, i can't open the (`http://<PLC_IP>/config/`)  website in browser, could you make some screen shot for it? i follow the setup3 as the website.  https://github.com/jisotalo/ads-client#setup-3---connecting-from-any-nodejs-supported-system-to-the-plc
sorry for the trouble.