jisotalo/ads-client #56: Is ads communication be possible in a browser?

mendeljacks, 2021-05-03:


jisotalo, 2021-05-03:
Hi! 

With this library it's only possible from Node.js. The reason is that communication is done with TCP which isn't available from browser. 

Perhaps in the future I will release a client that would work from browser using ADS web service or WCF (https://infosys.beckhoff.com/english.php?content=../content/1033/tcadswcf/html/tcadswcf.tcadsservice.endpoints.unsecwebhttp.html&id=6893539725153588511)

mendeljacks, 2021-05-03:
Ok good to know thanks

closed 2021-05-04 by jisotalo

der1igedon, 2022-09-28:
Thanks for the answer!
I would like to write a simple HMI using react for the frontend development and your library to communicate with TwinCat 3. How would it be possible to setup the communication if TCP is needed? Would be nice if you could give me hint how to set this up. Thanks!

jisotalo, 2022-09-29:
Hi

You can't use this library in react/frontend. You need a separate Node.js server backend running that handles the communication. 

You can then communicate between react and node.js using HTTP or websockets.

der1igedon, 2022-09-29:
Yes, thats what I thought. Do you know if there's already a solution by someone? Thanks!
A simple exmple of how to emit PLC variable updates through websockets (ex. with socket.io) would be nice.

jisotalo, 2022-09-29:
Unfortunately no. I have made that for few projects but those are private.

Just try to find a good socket.io example and then instead of example backend logic use ads-client and its subscribe functionality.

der1igedon, 2022-09-29:
Thanks a lot. I'll try it.