jisotalo/ads-client #62: Properties support

Hopperpop, 2021-05-18:
Would it be possible to support reading/writing properties who have a correct monitor pragma? (https://infosys.beckhoff.com/content/1033/tc3_plc_intro/2529692299.html)

Currently it returns always a fixed value depending on the type: INT:0, LREAL;NaN, UINT:35320, BOOL:false, ...
Beckhoff HMI also uses ADS and seems to be able to handle properties with a pragma.

jisotalo, 2021-05-18:
Hi!

I have used the monitoring pragma but never tried to read that using ADS. It probably needs some research.

Do you know if TwinCAT.Ads.dll for C# supports it?

Hopperpop, 2021-05-18:
At the moment I don't have any idea what needs to be done to get it working. And if other libraries support it. 
Indeed some research will be needed.

Hopperpop, 2021-06-04:
I found that adding `{attribute 'monitoring' := 'call'}` adds a rpc call for the setter and getter function (ex. "__setMyProp"). 
But trying to call these methods doesn't work as it fails when getting a handle:

```
ads-client createVariableHandle(): Creating handle to MAIN.fbTest#__setp_bool failed: ClientException: Response with ADS error received 
	at Client.callback (C:\Users\x\source\repos\node-red-contrib-ads-client\node_modules\ads-client\src\ads-client.js:6164:25) 
	at Client._onAdsCommandReceived (C:\Users\x\source\repos\node-red-contrib-ads-client\node_modules\ads-client\src\ads-client.js:6005:29) 
	at Client._onAmsTcpPacketReceived (C:\Users\x\source\repos\node-red-contrib-ads-client\node_modules\ads-client\src\ads-client.js:5866:29) 
	at Client._parseAmsTcpPacket (C:\Users\x\source\repos\node-red-contrib-ads-client\node_modules\ads-client\src\ads-client.js:5521:27)     
	at processImmediate (internal/timers.js:461:21) { sender: '_sendAdsCommand()', adsError: true, adsErrorInfo: { adsErrorType: 'ADS error', adsErrorCode: 1808, adsErrorStr: 'Symbol not found' }, metaData: null, errorTrace: [], getInnerException: null } +5ms
```

I found a related question here: [stackoverflow](https://stackoverflow.com/questions/64060431/reading-properties-from-twincat-function-blocks-using-c-sharp)
Bellow the type definition of a function block with set/get rpc calls.

```json
{
  "name": "",
  "type": "FB_Test",
  "size": 24,
  "offset": 0,
  "adsDataType": 65,
  "adsDataTypeStr": "ADST_BIGTYPE",
  "comment": "",
  "attributes": [],
  "rpcMethods": [
    {
      "version": 1,
      "vTableIndex": 3,
      "returnSize": 1,
      "returnAlignSize": 1,
      "reserved": 0,
      "returnTypeGuid": "95190718000000000000000000000030",
      "retunAdsDataType": 33,
      "retunAdsDataTypeStr": "ADST_BIT",
      "flags": 1,
      "flagsStr": [
        "DataType"
      ],
      "nameLength": 11,
      "returnTypeLength": 4,
      "commentLength": 0,
      "parameterCount": 0,
      "name": "__getp_bool",
      "returnType": "BOOL",
      "comment": "",
      "parameters": []
    },
    {
      "version": 1,
      "vTableIndex": 7,
      "returnSize": 0,
      "returnAlignSize": 0,
      "reserved": 0,
      "returnTypeGuid": "00000000000000000000000000000000",
      "retunAdsDataType": 0,
      "retunAdsDataTypeStr": "ADST_VOID",
      "flags": 1,
      "flagsStr": [
        "DataType"
      ],
      "nameLength": 11,
      "returnTypeLength": 0,
      "commentLength": 0,
      "parameterCount": 1,
      "name": "__setp_bool",
      "returnType": "",
      "comment": "",
      "parameters": [
        {
          "size": 1,
          "alignSize": 1,
          "adsDataType": 33,
          "adsDataTypeStr": "ADST_BIT",
          "flags": 1,
          "flagsStr": [
            "In"
          ],
          "reserved": 0,
          "typeGuid": "95190718000000000000000000000030",
          "lengthIsPara": 0,
          "nameLength": 6,
          "typeLength": 4,
          "commentLength": 0,
          "name": "p_bool",
          "type": "BOOL",
          "comment": ""
        }
      ]
    },
    {
      "version": 1,
      "vTableIndex": 12,
      "returnSize": 1,
      "returnAlignSize": 1,
      "reserved": 0,
      "returnTypeGuid": "95190718000000000000000000000030",
      "retunAdsDataType": 33,
      "retunAdsDataTypeStr": "ADST_BIT",
      "flags": 1,
      "flagsStr": [
        "DataType"
      ],
      "nameLength": 8,
      "returnTypeLength": 4,
      "commentLength": 0,
      "parameterCount": 0,
      "name": "mGetBool",
      "returnType": "BOOL",
      "comment": "",
      "parameters": []
    }
  ],
  "arrayData": [],
  "subItems": [
    {
      "name": "pbool",
      "type": "BOOL",
      "size": 1,
      "offset": 16,
      "adsDataType": 33,
      "adsDataTypeStr": "ADST_BIT",
      "comment": "",
      "attributes": [
        {
          "name": "DisplayMinValue",
          "value": "0"
        },
        {
          "name": "DisplayMaxValue",
          "value": "1"
        }
      ],
      "rpcMethods": [],
      "arrayData": [],
      "subItems": []
    },
    {
      "name": "p_bool",
      "type": "BOOL",
      "size": 1,
      "offset": 0,
      "adsDataType": 33,
      "adsDataTypeStr": "ADST_BIT",
      "comment": "",
      "attributes": [
        {
          "name": "DisplayMinValue",
          "value": "0"
        },
        {
          "name": "DisplayMaxValue",
          "value": "1"
        }
      ],
      "rpcMethods": [],
      "arrayData": [],
      "subItems": []
    }
  ]
}
```

jisotalo, 2021-06-11:
Thanks @Hopperpop!

I will check this and report

jisotalo, 2021-06-24:
Been quite busy with one side project (uses TC3, ads-client and React). However I tried to work on this now.

I created a STRING property:
[image]

I managed to use the following to read and write that property.
```js
//Reading property
let handle = await client.createVariableHandle('GVL_PropertyTest.FB.Str')
let value = await client.convertFromRaw(await client.readRawByHandle(handle), 'STRING(80)')
await client.deleteVariableHandle(handle)

console.log('Value at first: ', value)

//Writing property
handle = await client.createVariableHandle('GVL_PropertyTest.FB.Str')
await client.writeRawByHandle(handle, await client.convertToRaw('A new value!', 'STRING(80)'))
await client.deleteVariableHandle(handle)

console.log('New value was written!')

//Reading property again
handle = await client.createVariableHandle('GVL_PropertyTest.FB.Str')
value = await client.convertFromRaw(await client.readRawByHandle(handle), 'STRING(80)')
await client.deleteVariableHandle(handle)

console.log('Value after writing: ', value)
```

However, my PC crashed randomly when testing this...

jisotalo, 2021-06-30:
@Hopperpop have you tried the code above, does it work at your system?

Hopperpop, 2021-06-30:
I tried it and and seems to work correctly with LREAL. Code in the getter function is also executed.
For strings I also have crashes, even when only calling "readRawByHandle".

jisotalo, 2021-06-30:
Ok! Thanks for info. The crashing is a bit strange..

jisotalo, 2021-08-08:
I'm closing this issue for now. Will add somehting to the README later.

closed 2021-08-08 by jisotalo