jisotalo/ads-client #63: Uncaught Exception

Hopperpop, 2021-05-24:
I was trying to connect to a plc from a machine with TwinCat v3.1.4022.22, using the node-red wrapper (v1.10.4). 
For a still unknown reason the local router seems to drop the connection with "ECONNRESET". 
```
24 May 23:07:08 - [info] [ads-client-connection:287b52f4.ada39e] Connecting to 5.66.134.244.1.1:851...
  ads-client connect(): Starting to connect localhost:48898 +0ms
24 May 23:07:08 - [info] Started flows
  ads-client connect(): Socket connection established to localhost:48898 +7ms
  ads-client:details _registerAdsPort(): Registering an ADS port from ADS router localhost:48898 +0ms
  ads-client:raw-data IO out ------> 8 bytes : 0010020000000000 +0ms
  ads-client connect(): Socket connect failed: Error: read ECONNRESET
  ads-client     at TCP.onStreamRead (internal/stream_base_commons.js:209:20) {
  ads-client   errno: -4077,
  ads-client   code: 'ECONNRESET',
  ads-client   syscall: 'read'
  ads-client } +1ms
24 May 23:07:08 - [error] [ads-client-connection:287b52f4.ada39e] Connecting to 5.66.134.244.1.1:851 failed - Connection to localhost:48898 failed (socket error -4077) (read ECONNRESET)
24 May 23:07:08 - [info] [ads-client-connection:287b52f4.ada39e] Failed to connect 5.66.134.244.1.1:851 at startup
```
After playing around and setting the local AmsNetId and ADS port manually, I started to have Uncaught Exception. For node-red these are fatal and only way to recover is to run it in safe-mode (nodes not running at startup) and undo the manual settings.
```
24 May 22:47:39 - [info] [ads-client-connection:287b52f4.ada39e] Connecting to 5.66.134.244.1.1:851...
  ads-client connect(): Starting to connect localhost:48898 +0ms
24 May 22:47:39 - [info] Started flows
  ads-client connect(): Socket connection established to localhost:48898 +6ms
  ads-client:details _registerAdsPort(): Registering an ADS port from ADS router localhost:48898 +0ms
  ads-client _registerAdsPort(): Local AmsNetId and ADS port manually given so using 192.168.14.148.1.4:841 +0ms
  ads-client connect(): ADS port registered from router. We are 192.168.14.148.1.4:841 +1ms
  ads-client:details readSystemManagerState(): Reading device system manager state +0ms
  ads-client:details _sendAdsCommand(): Sending an ads command ReadState (0 bytes): { amsTcp: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD' }, ams: { targetAmsNetId: '5.66.134.244.1.1', targetAdsPort: 10000, sourceAmsNetId: '192.168.14.148.1.4', sourceAdsPort: 841, adsCommand: 4, adsCommandStr: 'ReadState', stateFlags: 4, stateFlagsStr: 'AdsCommand, Tcp, Request', dataLength: 0, errorCode: 0, invokeId: 0 }, ads: { rawData: <Buffer > } } +1ms
  ads-client:details _createAmsHeader(): AMS header created (32 bytes) +3ms
  ads-client:raw-data _createAmsHeader(): AMS header created: '054286f401011027c0a80e940104490304000400000000000000000000000000' +0ms
  ads-client:details _createAmsTcpHeader(): AMS/TCP header created (6 bytes) +2ms
  ads-client:raw-data _createAmsTcpHeader(): AMS/TCP header created: '000020000000' +1ms
  ads-client:details _createAmsTcpRequest(): AMS/TCP request created (38 bytes) +1ms
  ads-client:raw-data IO out ------> 38 bytes : 000020000000054286f401011027c0a80e940104490304000400000000000000000000000000 +2ms
24 May 22:47:39 - [red] Uncaught Exception:
24 May 22:47:39 - Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
```

Digging into the code I found:
```
        //Remove the socket events that were used only during connect()
        socket.removeAllListeners('error')
        socket.removeAllListeners('close')
        socket.removeAllListeners('end')
```
When disabling the `socket.removeAllListeners('error')` line, the error get caught and the socket gets destroyed. Here we get a new Uncaught Exception:
```
  ads-client:details _sendAdsCommand(): Sending an ads command ReadState (0 bytes): { amsTcp: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD' }, ams: { targetAmsNetId: '5.66.134.244.1.1', targetAdsPort: 10000, sourceAmsNetId: '192.168.14.148.1.4', sourceAdsPort: 841, adsCommand: 4, adsCommandStr: 'ReadState', stateFlags: 4, stateFlagsStr: 'AdsCommand, Tcp, Request', dataLength: 0, errorCode: 0, invokeId: 0 }, ads: { rawData: <Buffer > } } +2ms
  ads-client:details _createAmsHeader(): AMS header created (32 bytes) +2ms
  ads-client:raw-data _createAmsHeader(): AMS header created: '054286f401011027c0a80e940104490304000400000000000000000000000000' +0ms
  ads-client:details _createAmsTcpHeader(): AMS/TCP header created (6 bytes) +3ms
  ads-client:raw-data _createAmsTcpHeader(): AMS/TCP header created: '000020000000' +3ms
  ads-client:details _createAmsTcpRequest(): AMS/TCP request created (38 bytes) +3ms
  ads-client:raw-data IO out ------> 38 bytes : 000020000000054286f401011027c0a80e940104490304000400000000000000000000000000 +2ms
  ads-client:details but what about here +1ms
  ads-client connect(): Socket connect failed: Error: read ECONNRESET
  ads-client     at TCP.onStreamRead (internal/stream_base_commons.js:209:20) {
  ads-client   errno: -4077,
  ads-client   code: 'ECONNRESET',
  ads-client   syscall: 'read'
  ads-client } +13ms
24 May 22:51:56 - [error] [ads-client-connection:287b52f4.ada39e] Connecting to 5.66.134.244.1.1:851 failed - Connection to localhost:48898 failed (socket error -4077) (read ECONNRESET)
24 May 22:51:56 - [info] [ads-client-connection:287b52f4.ada39e] Failed to connect 5.66.134.244.1.1:851 at startup
  ads-client:details _sendAdsCommand(): Timeout for command ReadState with invokeId 0 - No response +2s
  ads-client readSystemManagerState(): Device system manager state read failed +2s
  ads-client disconnect(): Starting to close connection (force: false) +1ms
  ads-client unsubscribeAll(): Unsubscribing from all notifications +1ms
  ads-client unsubscribeAll(): Unsubscribed from 0 notifications +1ms
  ads-client _unsubscribeAllInternals(): Unsubscribing from all notifications +0ms
  ads-client _unsubscribeAllInternals(): Unsubscribed from 0 notifications +0ms
  ads-client:details _unregisterAdsPort(): Unregister ads port 841 from localhost:48898 +5ms
  ads-client _unregisterAdsPort(): Local AmsNetId and ADS port manually given so no need to unregister +1ms
  ads-client disconnect(): Connection closed successfully +2ms
  ads-client:details _unregisterAdsPort(): Socket closed +2ms
24 May 22:51:58 - [red] Uncaught Exception:
24 May 22:51:58 - TypeError: Cannot read property 'destroy' of null
    at C:\Users\Documents\node-red-contrib-ads-client\node_modules\ads-client\src\ads-client.js:3563:32
    at processTicksAndRejections (internal/process/task_queues.js:79:21)
```

In the end there are three issues:
- Uncaught Exception in ads-client. Why are the error/close/end listeners removed?
- Uncaught Exception in node-red during destroy.
- Unknown issue on my machine that cases all this. I tried on another machine with a more recent TwinCat version and didn't had an issues connecting.

jisotalo, 2021-05-30:
Hi Hopperpop!

Sorry for delay, again..

I checked the code and it seems that there are few lines where socket errors aren't catched.  The code you found is run in `connect()` method and it has its own event listeners for socket errors. After connection is successful (ADS port is registered), the error listener is removed. After that, a new one is created.

However, between those lines we have some code that reads system manager state and initializes internal subscriptions. The error event listener is added after those operations, which isn't correct. 

So lines `401...434`
```
try {
  //Try to read system manager state - If it's OK, connection is successful to the target
  await this.readSystemManagerState()
  _systemManagerStatePoller.call(this)

} catch (err) {
  try {
    await this.disconnect()
  } catch (err) {
    debug(`connect(): Reading target system manager failed -> Connection closed`)
  }
  return reject(new ClientException(this, 'connect()', `Connection failed: ${err.message}`, err))
}

try {
  await _reInitializeInternals.call(this)

} catch (err) { 
  if (this.settings.allowHalfOpen !== true) {
    try {
      await this.disconnect()
    } catch (err) {
      debug(`connect(): Connecting to target PLC runtime failed -> Connection closed`)
    }
    return reject(new ClientException(this, 'connect()', `Target and system manager found but couldn't connect to the PLC runtime (see setting allowHalfOpen): ${err.message}`, err))
  }

  //Todo: Redesign this
  if(this.metaData.systemManagerState.adsState !== ADS.ADS_STATE.Run)
    _console.call(this, `WARNING: Target is connected but not in RUN mode (mode: ${this.metaData.systemManagerState.adsStateStr}) - connecting to runtime (ADS port ${this.settings.targetAdsPort}) failed`)
  else
    _console.call(this, `WARNING: Target is connected but connecting to runtime (ADS port ${this.settings.targetAdsPort}) failed - Check the port number and that the target system state (${this.metaData.systemManagerState.adsStateStr}) is valid.`)
}
```

should be after these lines `436...443`.
```
//Listening connection lost events
this._internals.socketConnectionLostHandler = _onConnectionLost.bind(this, true)
socket.on('close', this._internals.socketConnectionLostHandler)
socket.on('end', this._internals.socketConnectionLostHandler)

//TODO: If socket error happens, should something to be done? Now probably close/end is called afterwards.
this._internals.socketErrorHandler = err => _console.call(this, `WARNING: Socket connection error: ${JSON.stringify(err)}`)
socket.on('error', this._internals.socketErrorHandler)
```

**I will fix that!**

Could you tell me more about your setup that isn't working? I have encountered ECONNRESET errors when sending something stupid to the server or registering faulty ADS port etc.

jisotalo, 2021-05-30:
Version 1.10.5 should no longer have this uncaught exception.

Hopperpop, 2021-06-01:
> Sorry for delay, again..

No need to apology. I fully understand, and appreciate all the effort!

> Could you tell me more about your setup that isn't working? I have encountered ECONNRESET errors when sending something stupid to the server or registering faulty ADS port etc.

- PLC: Cx8190 - Microsoft Windows Embedded Compact 7 - TwinCat runtime 3.1.4022.14
- Engineering (problem pc): Win10: TwinCat 3.1.4022.22 (at the time of reporting)
- Engineering 2 (no problems): Win10: TwinCat 3.1.4024.12

All connected to a normal local network, no VM/Dockers/... Connection was established in the router.  

I normally don't use the first engineering pc for TwinCat development. But I was recently using it to play around with different versions. I tried yesterday to match the engineering with the runtime on the plc (3.1.4022.14), but ran into problems even after trying a clean install. I would say at the moment the problem just comes from badly installed software. I will keep you posted if I know more, and maybe try to test the new version in the meantime.

jisotalo, 2021-06-01:
Have you tried to enable loopback? As it's not automatically enabled on older versions than 4024.5. Not sure though if it could cause this.
https://github.com/jisotalo/ads-client#enabling-localhost-support-on-twincat-3

Hopperpop, 2021-06-02:
Still no luck. I tried enabling the loopback, but that didn't change anything. I went into the rapid hole, trying to understand what's under the hood. However I'm starting to give up on that. Where did you get the information how to for example register the port? Is it by reverse engineering or am I missing some available resources?

I also tried to connect with the .NET adsclient, and that worked without an issue on the same machine. I wonder if that also connects to the local router with a TCP connection or uses another more direct way.

jisotalo, 2021-06-03:
Damn. Haven't seen this kind of a problem before. I have tested the library with 4020, 4022 and 4024 versions with success.

The port registering is same functionality as in the .NET library 5.x.

Have you tried to delete all routes and re-create them?

What is you ads-client setup. Are you using it like this with only two settings (target id and port):
```
const client = new ads.Client({
  targetAmsNetId: 'XXXXX',
  targetAdsPort: 851
})

client.connect()
  .then(res => {   
    console.log(`Connected to the ${res.targetAmsNetId}`)
    console.log(`Router assigned us AmsNetId ${res.localAmsNetId} and port ${res.localAdsPort}`)

    return client.disconnect()
  })
  .then(() => {
    console.log('Disconnected')
  })
  .catch(err => {
    console.log('Something failed:', err)
  })
```

Hopperpop, 2021-06-03:
I tried already many combinations and have found one that works. By using setup 3, opening tcp port on the plc, adding route on the plc and basically bypassing the local ads router.

[image]
[image]
[image]

All other combinations that tries to use the local ads router fails (like only providing targetAmsNetId and port).

Here I connected from a rPi to the problem pc ads router. Where it also drops the connection:
[image]

Here you can see that the ads router exist and listening on port 48898:
[image]

Maybe it's other software interfering, or a problem with the protocol. 

But it's a good test to test socket handling:
```
3 Jun 21:06:50 - [info] Stopping flows
3 Jun 21:06:50 - [info] [ads-client-connection:7ec07daa.d00aa4] Disconnecting from 5.66.134.244.1.1:851 and unsubscribing from all...
(node:17) UnhandledPromiseRejectionWarning: Error: This socket has been ended by the other party
    at Socket.writeAfterFIN [as write] (net.js:396:12)
    at Client._socketWrite (/data/node_modules/ads-client/src/ads-client.js:3794:26)
    at Promise (/data/node_modules/ads-client/src/ads-client.js:3606:18)
    at new Promise (<anonymous>)
    at Client._unregisterAdsPort (/data/node_modules/ads-client/src/ads-client.js:3552:10)
    at Promise (/data/node_modules/ads-client/src/ads-client.js:535:34)
    at process._tickCallback (internal/process/next_tick.js:68:7)
(node:17) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:17) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

jisotalo, 2021-06-04:
Hmm. This is quite strange.

Have you tried to run TwinCAT locally at that problematic computer and then connecting to localhost with ads-client? So instead of the real PLC, connecting the local runtime? That goes through the router too so just to check what happens.

I will add an error handler to _socketWrite function, thanks for reporting!

Hopperpop, 2021-06-09:
I haven't been able to run TwinCat locally on the problematic pc. It's always complaining about Hyper-V. Tried different things like isolated core, removing windows features,... But nothings helps.
It could be the issue causing this. Although if the router wasn't working, than I would suspect that other functionality also wouldn't work (.Net lib, online watch,...) 
I don't need this pc for TwinCat, it was just for testing. I will leave it like it is, maybe someday something will pops-up with a clue.

closed 2021-06-24 by jisotalo