jisotalo/ads-client #69: Getting Remote Routes via ADS

LaCocoRoco, 2021-06-29:
Hi this is no real "issue" more a "question". I am working on a windows environment. The local system runs a twincat service. There are several remote routes managed on this specific machine. Is there any way to get the remote Routes (AmsNetIds) from this machine with an ADS command? I couldn't realy find any documentation regarding the Ports and Index Group combinations.

jisotalo, 2021-06-29:
Hi! Your question was too interesting so I had to go deeper... 

I knew that that Tc2_Utilities library has function block [FB_EnumRouteEntry](https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclib_tc2_utilities/34984075.html&id=7637337685162133919) so I tested it and checked what is going on using ADS monitor.

I found the data:
[image]

It was quite trivial to parse. However there are 38 bytes that I couldn't find quickly. Perhaps transport mode, timeout time etc..

I updated ads-client to 1.11.0 so it has sendAdsCommand() available for use. You can try yourself if you like, update the ads-client and check the following gist:
https://gist.github.com/jisotalo/cfcc9e4935dfc6f391667cad2cdcb6c6

Result is something like:
```
[
  {
    amsNetId: '10.255.2.15.1.1',
    _unknown: <Buffer 01 00 00 00 00 00 60 ea 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 03 00 00 00 00 00 00 00>,
    address: '192.168.5.153',
    name: 'XP'
  },
  {
    amsNetId: '192.168.5.124.1.2',
    _unknown: <Buffer 01 00 00 00 00 00 60 ea 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 03 00 00 00 00 00 00 00>,
    address: '192.168.5.124',
    name: 'pi'
  },
  {
    amsNetId: '5.61.99.148.1.1',
    _unknown: <Buffer 01 00 00 00 00 00 60 ea 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0e 00 00 00 0a 00 00 00 00 00 00 00>,
    address: '192.168.5.150',
    name: 'CX-3D6394'
  }
]
```

Note that this is not anything 100% working, just coding for fun :)

LaCocoRoco, 2021-06-29:
That is perfect! Especially now i can somehow understand how you got this far with your project! This is more and by far faster then anything i expected. It looks very promissing but cant say when i have time to test this. Thank you very mutch for the help, i will close this as solved.

closed 2021-06-29 by LaCocoRoco

jisotalo, 2021-06-30:
Haha, thanks! Not always this fast, just had some spare time and you had an interesting question!