jisotalo/ads-client #82: ads-client in a Docker

hbTecAG, 2022-01-25:
Hi

I have a problem when running Node-RED in a docker.
Communication then only works if I run the container with --network host . In other words, the entire host network into the container route.
https://docs.docker.com/network/network-tutorial-host/
Since this isn't the goal of Docker, I'm wondering what ports I need to route for the configuration to work.

Raspberry Pi 4 with docker container
CX9020 TwinCAT 2.11 Runtime

I have already tried the following ports:
- 48898/tcp + 48898/udp (localTcpPort)
- 48899/tcp + 48899/udp (found on Beckhoff site)
- 801/tcp + 801/udp (TwinCAT 2.11 Runtime)
- 32767/tcp + 32767/udp (localAdsPort)

[image]

jisotalo, 2022-01-26:
In my tests and experience, the 48898 TCP port is the only one that is required.

It's important to understand that the 48898 is the port that the **PLC/TwinCAT router is listening to.** However the ads-client itself uses next available/random TCP port to connect from, unless set excplicitly to port with `localTcpPort` setting, as you have.

So if you set `localTcpPort `to like 34567, and you connect to target PLC's port 48898, you need to allow communication from docker 34567 -> PLC 48898. If you don't set `localTcpPort`, you need to allow all communication from docker -> PLC 48898.

Hope this message makes any sense.

hbTecAG, 2022-01-26:
Hi, 

Thanks for your Answer. This make sense, but it didn't work. 
if i route the Port 48898/tcp (`-p 48898:48898`) in to the Container, it doesn't work. 

Maybe it is different what the `-p` Parameter does in comparison to `--network host` Parameter..

Interestingly, the [node-red-contrib-ads](https://github.com/PLCHome/node-red-contrib-ads) node works without port forwarding (Nor 48898). The connection is established as soon as the route is entered on the PLC.
Is there a difference in the connection in the Lib?

jisotalo, 2022-01-26:
Interesting. What happens if you set localAdsPort and localAddress empty?

Of course these are totally different libraries. Could you share all the settings from both?

hbTecAG, 2022-01-26:
Sure:

[image]
[image]

IP's are not the same, only an example how it works.

If i set localAdsPort and localAddress empty, it doesn't work. It also doesn't work in `--network host` mode..
If i only leave localAddress empty, it works. Seems to be the localAdsPort is required..?

jisotalo, 2022-01-26:
Sorry.. I mean localTcpPort, not ads port. The ADS port is required as you are using Raspberry Pi without AMS router.

hbTecAG, 2022-01-26:
same result: 

[image]

jisotalo, 2022-01-26:
So both localAddress and localTcpPort are now empty?`

EADDRNOTAVAIL means that it can't start a connection from local interface 10.1.47.50. 
https://www.google.com/search?q=eaddrnotavail+site:stackoverflow.com

hbTecAG, 2022-01-26:
if both are empty, it seems to work! 

Maybe, docker has a problem when localAddress is set, because of every container has it own ip.. ?
if these feelds are empty, it works, even without any port routing. 
Great!

Maybe you can add this in your description..?
Thanks alot!

jisotalo, 2022-01-26:
Glad to hear! :)

The localAddress can be empty almost always. Only reason I can image is that you have many interfaces (like ethernet, wifi, etc) and you need to provide manually which one to use.

So the problem indeed was that it tried to connect using 10.1.47.50 however the was no such interface in your docker container.

I'll see if I could add this info somewhere in FAQ!

closed 2022-01-26 by jisotalo