jisotalo/ads-client #83: Problems with TOD / TIME_OF_DAY datatype

schgab, 2022-01-31:
I use the `readSymbol` function and reading a structure containing a TOD variable or simply reading a TOD throws an exception: 
`Uncaught (in promise) ClientException: Reading data type info for TOD failed`
[localhost-1643632672066.log](https://github.com/jisotalo/ads-client/files/7970689/localhost-1643632672066.log)

jisotalo, 2022-01-31:
Hi!

I tested it on my local PC running TwinCAT 4024.10 and ads-client 1.12.1. Reading `TOD `and `TIME_OF_DAY` both worked fine.

What versions are you using?

PLC:
```
//GVL_Test
testTOD : TOD := TOD#15:30;
```

Javascript:
```
const data = await client.readSymbol('GVL_Test.testTOD')
console.log(data)
```

Console output:
```
{
  value: 55800000,
  type: {
    name: '',
    type: 'TIME_OF_DAY',
    size: 4,
    offset: 0,
    adsDataType: 19,
    adsDataTypeStr: 'ADST_UINT32',
    comment: '',
    attributes: [],
    rpcMethods: [],
    arrayData: [],
    subItems: []
  },
  symbol: {
    indexGroup: 16448,
    indexOffset: 455700,
    size: 4,
    adsDataType: 19,
    adsDataTypeStr: 'ADST_UINT32',
    flags: 8,
    flagsStr: [ 'TypeGuid' ],
    arrayDimension: 0,
    nameLength: 16,
    typeLength: 11,
    commentLength: 0,
    name: 'GVL_Test.testTOD',
    type: 'TIME_OF_DAY',
    comment: '',
    arrayData: [],
    typeGuid: '9519071800000000000000000000004b',
    attributes: [],
    reserved: <Buffer 00 00 00 00>
  }
}
```

schgab, 2022-02-01:
> What versions are you using?

I am using Twincat 2 on  a CX9020:
TC CE Build: 354
TC Build: 2258

In the PLC I have a a function block as follows: 
```
FUNCTION_BLOCK GetSystemTime_FB
VAR_OUTPUT
	dt_DateTime: DT; 			
	t_Time: TIME;				
	tod_TimeOFDay: TOD;			
	B_DayOfWeek: BYTE;			
	w_TaskCyclus: WORD;			
	st_Time: TIMESTRUCT;		
END_VAR
VAR
	GetTime: NT_GetTime;
	GetTaskCyclus: GETTASKTIME;
	d_Task_High_Before: DWORD;
	d_Task_Low_Before: DWORD;
	d_Task_High: DWORD;
	d_Task_Low: DWORD;
END_VAR
```
I have an instance of that function block:
`SystemTime:									GetSystemTime_FB;`

And the following Javascript yields the same error:
```
(async () => {

   const client = new ads.Client({
    targetAmsNetId: ams,
    targetAdsPort: 801,
   })

    await client.connect()
    console.log(await client.readSymbol(".systemtime"))   
    await client.disconnect()
   
})()
```
`ClientException: Reading data type info for TOD failed`

jisotalo, 2022-02-01:
Ok thanks! The reason is that in TC2 the PLC doesn't provide information for basic data types, however this library should handle it. Does it work with `TIME_OF_DAY` instead? The reason probably is that I have forgotten to add `TOD` to base data types (`TIME_OF_DAY`  is there).

Please edit line 1014 from ads-client-ads.js:
https://github.com/jisotalo/ads-client/blob/master/src/ads-client-ads.js#L1014

`name: ['UDINT', 'DWORD', 'TIME', 'TIME_OF_DAY', 'BITARR32', 'UINT32'],`
-->
`name: ['UDINT', 'DWORD', 'TIME', 'TIME_OF_DAY', 'TOD', 'BITARR32', 'UINT32'],`

Does it work now?

schgab, 2022-02-01:
It works!

jisotalo, 2022-02-01:
Great! I'll fix it in the library too.

jisotalo, 2022-02-12:
Fixed in v. 1.12.2

closed 2022-02-12 by jisotalo