jisotalo/ads-client #84: Uncaught exception ECONNRESET after restarting PLC

dotKokott, 2022-02-03:
Hello,

I would like to be able to brace for a power outage of the PLC (including the switch that connects the PLC to the computer running node-ads).

When I turn off the power cabinet I get a connectionLost which is correct. However when it comes back online I get a
```
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
Emitted 'error' event on Socket instance at:
    at emitErrorNT (internal/streams/destroy.js:106:8)
    at emitErrorCloseNT (internal/streams/destroy.js:74:3)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  errno: -4077,
  code: 'ECONNRESET',
  syscall: 'read'
}
```

which I assume happens when it is trying to reconnect.

I am not sure how to catch this error. I would be fine with catching this moment and attempting the reconnection myself, however this crashes my whole application.

I already upgraded to the latest version 1.5.1

jisotalo, 2022-02-04:
Hi!

Are you sure you are using the newest version? Newest is **1.12.1** (not 1.5.1) and 1.12.0 incluled lots of optimizations to connection handling.

jisotalo, 2022-02-09:
@dotKokott thanks for reporting! I assume it worked with newest version?

dotKokott, 2022-02-09:
Yes it works now! thank you very much :)