jisotalo/ads-client #85: Ads notification received but it has unknown notificationHandle (39). Use unsubscribe() to save resources.

dotKokott, 2022-02-09:
Hello,

after upgrading I started receiving a lot of `Ads notification received but it has unknown notificationHandle (39). Use unsubscribe() to save resources.` messages.

How exactly can a notification handle get lost and how could I debug this? Is it possible to find out what the initial symbolName for that subscription was?

Thank you!

jisotalo, 2022-02-11:
Hi!

Basically how this could happen (I think) is that you have active connection and you have subscribed to different variables. Then the connection goes down, you kill the process or something else unexpected. The PLC does not know that and it keeps sending those notifications. If you close the client using `disconnect()`, it automatically unsubscribes all. But if the connection "just loses", the plc continues sending.

So the only way to stop notifications from sending is to restart PLC or to manually send unsubscribe command with the handle.

closed 2022-02-24 by jisotalo

dotKokott, 2022-07-23:
Hey I finally found the issue and thought it would be worth mentioning. So even after cleanly disconnecting every time I started getting those messages about unknown notificationHandle, even after clean restarts and they would start accumulating.

This was the problem:
We have a setup where we have the ads-client running on the same computer as the TwinCAT XAE shell which are both connecting to the remote PLC. However instead of using the simple connection config we were using the one were we specifiy the AMS router:
```
const client = new ads.Client({
  localAmsNetId: '192.168.1.10.1.1',  //Can be anything but needs to be in PLC StaticRoutes.xml file
  localAdsPort: 32750,                //Can be anything that is not used
  targetAmsNetId: '192.168.1.120.1.1',
  targetAdsPort: 851,
  routerAddress: '192.168.1.120',     //PLC ip address
  routerTcpPort: 48898                //PLC needs to have this port opened. Test disabling all firewalls if problems
})
```

For the router address we used the PLC's address. This seems to be actually illegal as only one entity (router or client) should be connecting from one IP address. What seemed to be happening is that we were receiving subscription updates to our ads-client that were meant for the XAE shell that was running at the same time.

The moment we switched for the simple connection config of just specifying the targetAmsNetId (and let the local router figure out the rest) all problems we had regarding connection issues with both ads-client and shell + unknown notificationHandle messages disappeared!

jisotalo, 2022-08-02:
Hi @dotKokott!

Sorry for late reply. I read your message on the same day but totally forgot to answer.

Thanks for reporting this. It seems that when using a router you can have as many connections as you like. Otherwise just one or so, however I'm not sure how it really was. I investigated that when I started working on this library but now totally forgot...

I will add this info to the FAQ! Certainly interesting that you found reason for that unknown handle problem, I have never seen that yet myself.