jisotalo/ads-client #91: Incorrect parsing of ARRAY of ARRAY syntax

isaac-nls, 2022-05-04:
Hi,

I'm running into a parsing issue with ads-client not parsing ARRAY of ARRAY syntax correctly. This is an alternative syntax for multidimensional arrays. For reference: [Beckhoff example](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/8825255691.html&id=). Specifically, the behaviour I'm getting is that ads-client produces a single-dimensional array of the base type.

For example, in TwinCAT I make the following declaration:
```
PROGRAM MAIN
VAR
	arrNormalArray			: ARRAY[1..3] OF BOOL;
	arrMultiArray			: ARRAY[1..3, 1..2] OF BOOL;
	arrOfArrays			: ARRAY[1..3] OF ARRAY[1..2] OF BOOL;
END_VAR
```

Reading this using ads-client, I get the following result:
```
ARRAY [1..3] OF BOOL = [ false, false, false ]
ARRAY [1..3,1..2] OF BOOL = [ [ false, false ], [ false, false ], [ false, false ] ]
ARRAY [1..3] OF ARRAY [1..2] OF BOOL = [ false, false, false ] // Incorrect, should be  [ [ false, false ], [ false, false ], [ false, false ] ]
```

My full JS code for producing this is below
```
const ads = require('ads-client');

let a = new ads.Client({
    localAmsNetId: "192.168.1.10.1.1", // Can be anything but needs to be in PLC StaticRoutes.xml file
    targetAmsNetId: "127.0.0.1.1.1",
    targetAdsPort: 851,
    disableBigInt: true, // always disable bigint for serialising
    objectifyEnumerations: false // don't turn ENUMs into objects
});

const f = async () => {
    await a.connect();

    let normal = await a.readSymbol("MAIN.arrNormalArray");
    console.log(normal.symbol.type, "=", normal.value);

    let multi = await a.readSymbol("MAIN.arrMultiArray");
    console.log(multi.symbol.type, "=", multi.value);

    let arrOfArr = await a.readSymbol("MAIN.arrOfArrays");
    console.log(arrOfArr.symbol.type, "=", arrOfArr.value, "// incorrect!");
}

f();
```

Thank you for taking the time to read and for making this library available, it's been fantastic aside from this little bug.

Isaac

jisotalo, 2022-05-05:
Hi Isaac!

Thanks for reporting this. It certainly should work but it seems not, I will check this in a few days.

Glad to hear that the library has been useful!

isaac-nls, 2022-05-06:
Hi Jussi,

No worries at all, thanks for checking it out.

Thanks!
Isaac

jisotalo, 2022-05-17:
Hi @isaac-nls!

I fixed the problem in version 1.13.2. At least in my tests everything works now fine. Can you confirm?

Sorry for the delay!

isaac-nls, 2022-05-18:
Hi @jisotalo,

Fantastic! I've run my tests above and tested in my broader project and it seems to be fixed on my end too. I even tested it on this abomination below and it worked:
```arr4DHybrid				: ARRAY[1..4, 1..3] OF ARRAY[1..5] OF ARRAY[1..3] OF BOOL;```

Again, thank you very much for your help in resolving the issue, much appreciated!

Thanks!
Isaac

closed 2022-05-18 by isaac-nls

jisotalo, 2022-05-18:
Awesome! Thanks for reporting.