jisotalo/ads-client #93: ads-client on twin cat 2 PLC support

martinschki, 2022-06-05:
Hallo,
excuse me if this is not the right way to contact you.
I'm currently using a hard PLC CX9010-1001 which I'm programming with twin cat 2 in connection with node-red-contrib-ads (1.1.30) which is working fine.

Since that package is no longer maintained and my house depends on that ADS connection, I'd like to switch to your client.

I'm basically using "ADS Notification" and "ADS output" to read and write variables via symbol names to further use and manipulate in node-red.

I have read in the documentation that it is not tested with a setup like mine so my question is, would you support me if I'd test and then use it in my setup?
Thank you

jisotalo, 2022-06-06:
Hi!

This library has been used in many TwinCAT 2 systems by others. I haven't tried it myself.

If you want, please try the node-red library that is based on this and tell your feedback.
https://flows.nodered.org/node/node-red-contrib-ads-client

jisotalo, 2022-06-13:
Any news on this?

martinschki, 2022-06-18:
Sorry, I'm a little bit tight on time right now...will try soon, as I want to switch to a supported lib!

jisotalo, 2022-07-20:
Closing this for now

closed 2022-07-20 by jisotalo