jisotalo/ads-client #98: Writing from PLC to Node Application

Phil-Davies-Github, 2022-09-22:
Hi, thanks for the library. 

As the node client application has no router attached to it I assume that only read and write from the node client is possible.
Is it possible to read the state of the node client and then write to from the PLC? 

Also do you have the typescript def file you could share please?

Thanks Phil

jisotalo, 2022-09-22:
If you want to write from PLC to Node.js, you need to run ads-server (https://github.com/jisotalo/ads-server).

This project has no typescript definitions yet. That ads-server project does thought.

closed 2022-09-22 by Phil-Davies-Github

jisotalo, 2022-09-23:
Thanks! 

I'm actually planning to rewrite the library to Typescript someday. Let's see..