jisotalo/ads-client #99: Connection issues

RikMe, 2022-10-21:
PLC: 192.168.11.3
AMSnetID PLC: 5.50.37.33.1.1
Twincat 2

Raspberry Pi : 192.168.11.2

Route has been created on PLC. Settings should be according tot tutorials

Error : "Failed to connect 5.50.37.33.1.1:801 at startup: ClientException: Connection failed: Device system manager state read failed - ADS error -1 (Timeout - no response in 6000 ms)"

[image]
[image]
[image]

jisotalo, 2022-10-25:
Can you set debugging level to 3 and copy paste the console output here?

RikMe, 2022-10-25:
[image]
[image]

jisotalo, 2022-10-25:
I mean from Node.js side (terminal).

RikMe, 2022-10-25:
Not quite sure how to pull these up. i'm using the function blocks in NodeRed

jisotalo, 2022-10-25:
Did you start node-red from a command line or terminal? If yes, then you will see it there.
If not, it's most probably logged somewhere else.

See 
https://stackoverflow.com/questions/39243198/where-to-find-console-in-node-red
https://nodered.org/docs/user-guide/runtime/logging

The latter has a note:
`Note : if you are running as a service on the Raspberry Pi using our install script, you can use the node-red-log command to view the service log.`

RikMe, 2022-10-25:
[image]

RikMe, 2022-10-25:
i see targetAdsPort : 1000. Should this not be 801 on TC2 ? is configured as 801 in settings.

jisotalo, 2022-10-27:
Unfortunately I don't know why it isn't working.  Are you sure that you don't have multiple connections already? See https://github.com/jisotalo/ads-client/issues/47

10000 is the port for system manager so that is correct.

jisotalo, 2022-12-08:
Any news on this?

closed 2022-12-10 by jisotalo

RikMe, 2022-12-11:
No, I tried multiple other settings but no effect I'm afraid.