jisotalo/ads-client #103: We have discovered a crash in the unsubscribe all method.

PieterJanDevolder, 2023-02-09:
[image]

jisotalo, 2023-02-09:
Hi. 

Thanks for reporting. Does this happen every time you call it?

PieterJanDevolder, 2023-02-09:
> Hi.
> 
> Thanks for reporting. Does this happen every time you call it?

No this happens after some time, hard to simulate this situation.

jisotalo, 2023-02-09:
Could you provide an example code?

PieterJanDevolder, 2023-02-10:
> Could you provide an example code?

i've sent an email

jisotalo, 2023-03-01:
This was discussed in email with @PieterJanDevolder:

> I tried to recreate this issue and only way I could manage to do it was calling unsubcribeAll() multiple times at the same time. So another promise was already unsubscribing when the next one tried to unsubscribe again.
> 
> So one possible reason could be that you code is doing that twice or ads-client is already doing it itself as it has noticed the connection loss. The client will reconnect and unsubscribe&resubscribe automatically if connection gets lost. I definetely need to add some null-checking to the method in the future.

So the reason might be that the subscription is already deleted. There is no null check in error handler at `ads-client.js` row 1227 to see if subscription exists:
https://github.com/jisotalo/ads-client/blob/master/src/ads-client.js#L1227

There should be a check just like in unsubscribe():
https://github.com/jisotalo/ads-client/blob/master/src/ads-client.js#L1184

The original reason is still a mystery. Maybe the ads-client is reconnecting at the same time and has just unsubscribed already.

I will close this issue when the null check is added.

jisotalo, 2023-05-02:
Fixed in 1.14.2

closed 2023-05-02 by jisotalo