jisotalo/ads-client #104: Proposal ReadSymbolInfo SumCommand

aronneGrinton, 2023-02-10:
Hi,

What about to have a sumComman to read multiple symbolInfo in one shot?

Something like:

`const plcSymbolNames = ["MAIN.i_Integer1", "MAIN.i_Integer2", "MAIN.i_Integer3", "Main.FB_Test.b_Boolean"];`
`const totalStringLength= plcSymbolNames .reduce((total, target) => total + target.length + 1, 0)`
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``
``

jisotalo, 2023-02-13:
Hi!

I will check this in few days, but it could be doable. 

One way would be to use `readAndCacheSymbols()` once in the beginning and the get the symbol info from local cache using `getSymbolInfo()`.

aronneGrinton, 2023-02-14:
I tried this but not all the PLC varaibles are retrived. Maybe I'm wrong, I will make other tests.

aronneGrinton, 2023-03-07:
Hi!
I made some tests and in my case `readAndCacheSymbols()` get "specific informations" only for global variables.
If a PLC variables is declared inside a FB instantiated in a PRG, `readAndCacheSymbols()` does not ge "specific informations" on it, since it retrives information only of the PRG.
So if I use `getSymbolInfo()`, the info are requested to PLC and not to the cached symbols.
For this reason if yuo have a lot of "nested symbols", better to use a sumCommand in order to avoid the overflow of the Ads router mailbox.

closed 2023-05-02 by jisotalo