jisotalo/ads-client #107: exported code of matlab has type MatSim.DINT

JarmoCluyse, 2023-03-24:
the interface communicates with matlab/simulink generated code.
We never had an issue before, 
with the new code generator if we generate a int32 it changes to this type and the package crashes with this error:

\node_modules\ads-client\src\ads-client-ads.js:956
      throw new Error(`Error: Base type ${name} not found from BaseDataTypes - If this should be found, report an issue`)

Error: Error: Base type MatSim.DINT not found from BaseDataTypes - If this should be found, report an issue

jisotalo, 2023-04-13:
Hi @JarmoCluyse!

I had somehow missed this issue, sorry.

Could you elaborate more? Any code example? It seems that instead of the basic type `DINT` the Simulink code is using MatSim.DINT or something. I have never used the converter myself.

JarmoCluyse, 2023-04-24:
This is indeed what happens, The type DINT get's replaced by the type MatSim.DINT
the problem is I only write the tooling and I don't have the matlab licence

jisotalo, 2023-05-02:
Could you provide an example of the matlab-generated PLC code?

JarmoCluyse, 2024-01-02:
Hi,
we eventually found the issue,
it was a bug in the matlab compiler, this has been fixed in newer versions

closed 2024-01-02 by JarmoCluyse

jisotalo, 2024-01-03:
Thank you for confirmation, that's good!