jisotalo/ads-client #108: Optional Ams_NetId

Hopperpop, 2023-04-05:
To read CoE data from an EtherCat master over ads, you need to use another AMS_NetId than the one you use for setting up the route. 
Standard the first EtherCat master is assigned x.x.x.x.2.1. The raw read-write-read/write commands should have an additional optional Ams_NetId argument to make it possible to send commands to other ads-devices like an EtherCat master.

Does this make sense? Or is this already possible in another way?

jisotalo, 2023-04-13:
Hi @Hopperpop!

That certainly would be one option. I remember connecting to EtherCAT devices by creating the route manually to StaticRoutes.xml. I need to test this.

jisotalo, 2023-05-02:
I will look into this during development of v2 version.

jisotalo, 2024-07-13:
In version 2, every method has optional parameter `targetOpts`, (`targetOpts: Partial<AmsAddress> = {}`) that allows providing amsNetId or adsPort. This way it's easily possible to connect different ports with the same connection.

closed 2024-07-13 by jisotalo