jisotalo/ads-client #109: _parseDataType: Incorrect parsing of flags

mycroes, 2023-05-02:
In the following excerpt from `_parseDataType`:
```js
  //20..23 Offset
  dataType.offset = data.readUInt32LE(pos)
  pos += 4

  //24..27 ADS data type (AdsDataType)
  dataType.adsDataType = data.readUInt32LE(pos)
  dataType.adsDataTypeStr = ADS.ADS_DATA_TYPES.toString(dataType.adsDataType)
  pos += 4

  //27..30 Flags (AdsDataTypeFlags)
  dataType.flags = data.readUInt16LE(pos)
  dataType.flagsStr = ADS.ADS_DATA_TYPE_FLAGS.toStringArray(dataType.flags)
  pos += 4

  //31..32 Name length
  dataType.nameLength = data.readUInt16LE(pos)
  pos += 2

  //33..34 Type length
  dataType.typeLength = data.readUInt16LE(pos)
  pos += 2
```

The offsets in the comments are off by one starting at Flags (should have been `28..31`) and flags is parsed using `readUInt16LE`, while `ADS_DATA_TYPE_FLAGS` actually contains values > 16 bits. `pos` is incremented by 4, so it's mostly comments that are messed up.

Noticed this while expanding the [AdsClient library](https://github.com/mycroes/AdsClient), so big thank you for the detailed implementation.

jisotalo, 2023-05-02:
Hi @mycroes!

Thank you for reporting this, it indeed seems to be a bug! Also by looking at [`ADS_DATA_TYPE_FLAGS`](https://github.com/jisotalo/ads-client/blob/master/src/ads-client-ads.js#L684) we can see that these values go over 16 bits.

I will merge your pull request, thanks.

Your library looks nice, keep up the good work!

closed 2023-05-02 by jisotalo

mycroes, 2023-05-02:
You're welcome! It's actually not my library, as in, I'm not the original author. However, I've recently acquired ownership of the GitHub repository and am in the process of expanding and refactoring the library to make it well-suited and performant for professional use.

I fixed at least one issue by using your code as reference and I'm making use of a lot of the other code to implement (or extend) things like retrieving symbols and datatypes. I still have to tackle updates to symbols though, but I guess your library provides enough information to handle that as well. So again, a big thanks for the detailed implementation, I'll put it to good use!

jisotalo, 2023-05-02:
Sounds good! Let me know if you have any issues by email, for example. It's always interesting to help.

commit 97370c11 (2023-05-02): Merge pull request #110 from mycroes/data-type-flags

Data type flags (fixes #109)