jisotalo/ads-client #112: Is it possible to access the library created with TwinCAT?

runtimevic, 2023-05-11:
Is it possible to access the library created with TwinCAT?
 when a library *.library is created, can it be accessed?

jisotalo, 2023-05-16:
I'm not sure what do you mean.

If you have a library and then add it to a PLC project, you can read/write data from the library GVLs, programs etc.

runtimevic, 2023-05-16:
> I'm not sure what do you mean.
> 
> If you have a library and then add it to a PLC project, you can read/write data from the library GVLs, programs etc.

Hello @jisotalo,
Let me explain better, I would like to carry out unit tests of a *.library,
- I do not want to install the *.library in any project, can this be done? 
- Another option could be to just install it in a project and make a reference in the project but not use it, having to call its FB in the project. Would this other option be viable?  
- The last option to carry out unit tests of a *.library I have to use it in the TwinCAT project by calling its FB and then it is the only way to carry out unit tests of developed libraries, 
I hope it is better understood.

jisotalo, 2023-05-21:
Hi @runtimevic 

- I do not want to install the *.library in any project, can this be done?
  - No, because the library needs to be running in a PLC runtime

- Another option could be to just install it in a project and make a reference in the project but not use it, having to call its FB in the project. Would this other option be viable?
  - That's totally possible, however the FBs need to be instantiated somewhere (for example GVL inside library)

closed 2023-06-19 by jisotalo