jisotalo/ads-client #114: Read symbol value on BC9050

ptorrent, 2023-08-18:
Hello there,

Thanks for your awesome job !

I've a question on BC9050 :

This is my global variable:

VAR_GLOBAL
                                                                                                              Index Group      Index Offset
                b31_DO_01        AT%QX0.0 : BOOL;                          16#F031           0*8+0
                b31_DO_02        AT%QX0.1 : BOOL;                          16#F031           0*8+1
                b31_DV_01        AT%MX10.0 : BOOL;                      16#4021            10*8+0
                b31_DV_02        AT%MX10.1 : BOOL;                      16#4021            10*8+1
                b31_DV_03        AT%MX10.2 : BOOL;                      16#4021            10*8+2
                i31_AV_01          AT%MB0 : INT;                                 16#4020            0
                i31_AV_02          AT%MB2 : INT;                                 16#4020            2
                i31_AV_03          AT%MB4 : INT;                                 16#4020            4

END_VAR

```
const client = new ads.Client({
  localAmsNetId: '192.168.1.110.1.1',  
  localAdsPort: 32750,      
  targetAmsNetId: '192.168.1.231.1.1',
  targetAdsPort: 800,
  allowHalfOpen : true,
 readAndCacheSymbols : true,
 routerAddress: '192.168.1.231',   
 routerTcpPort: 48898                

})

client.connect().then(async function(){
	console.error('connect')

	console.error(client.metaData.symbols)

	 var result = await client.readRaw(0x4020,0,2)
	console.log(result)
	var value = await client.convertFromRaw(result, 'INT')
	console.log(value) /

	result = await client.readRaw(0x4020,2,2)
	console.log(result)
	value = await client.convertFromRaw(result, 'INT')
	console.log(value) 

	const res = await client.readSymbol('b31_DO_02')
	console.error(res)
})

```

I'm able to get correct value on 0x4020,0,2  AND 0x4020,2,2. But i'm not able to read symbol "b31_DO_02". readAndCacheSymbols  seems not listing symbols but I think it's normal ?

Is there a way to be able to read symbol from that device ?

Thanks a lot for your support !

jisotalo, 2023-08-22:
Hi, thanks!

First, it might be that the BC9050 has no symbol information available. At least I have that kind of feeling as it's very simple device. If that is so, then only way is to read using raw addresses.

However, there is at least one error in your code. As the BC9050 is TwinCAT 2, the variable under a GVL needs to have dot (.) beforehand. See https://github.com/jisotalo/ads-client#important-things-to-know-when-using-with-twincat-2

So instead of 
`var symb = await client.readSymbol('b31_DO_02')`
use
`var symb = await client.readSymbol('.b31_DO_02')` (notice the dot).

However, it might still not work because of the BC device.

ptorrent, 2023-08-30:
Thanks a lot for you answer !

I was able to get value with raw data on BC device.

have a nice day !

closed 2023-09-23 by jisotalo