jisotalo/ads-client #116: BK9050

ptorrent, 2023-09-04:
Hello,

I'm trying to connect to BK9050.

Without bareClient actived this is what I got during the connecting process:

```
04-09-2023 15:23:12.862 > error > ads_24 > { ClientException: Connection failed: Device system manager state read failed
04-09-2023 15:23:12.862 > error > ads_24 >     at Socket.socket.once (/node_modules/ads-client/src/ads-client.js:3425:25)
04-09-2023 15:23:12.862 > error > ads_24 >     at <anonymous>
04-09-2023 15:23:12.862 > error > ads_24 >   name: 'ClientException',
04-09-2023 15:23:12.862 > error > ads_24 >   sender: '_connect()',
04-09-2023 15:23:12.862 > error > ads_24 >   adsError: true,
04-09-2023 15:23:12.862 > error > ads_24 >   adsErrorInfo:
04-09-2023 15:23:12.862 > error > ads_24 >    { adsErrorType: 'AMS error',
04-09-2023 15:23:12.862 > error > ads_24 >      adsErrorCode: 8,
04-09-2023 15:23:12.862 > error > ads_24 >      adsErrorStr: 'Unknown command ID' },
04-09-2023 15:23:12.862 > error > ads_24 >   metaData: null,
04-09-2023 15:23:12.862 > error > ads_24 >   errorTrace:
04-09-2023 15:23:12.862 > error > ads_24 >    [ 'readSystemManagerState(): Device system manager state read failed',
04-09-2023 15:23:12.862 > error > ads_24 >      '_sendAdsCommand(): Response with AMS error received' ],
04-09-2023 15:23:12.862 > error > ads_24 >   getInnerException: [Function] }
04-09-2023 15:23:12.921 > error > ads_24 > { ClientException: Connection failed: Device system manager state read failed
04-09-2023 15:23:12.921 > error > ads_24 >     at Socket.socket.once (/weble/supervision/node_modules/ads-client/src/ads-client.js:3425:25)
04-09-2023 15:23:12.921 > error > ads_24 >     at <anonymous>
04-09-2023 15:23:12.921 > error > ads_24 >   name: 'ClientException',
04-09-2023 15:23:12.921 > error > ads_24 >   sender: '_connect()',
04-09-2023 15:23:12.921 > error > ads_24 >   adsError: true,
04-09-2023 15:23:12.921 > error > ads_24 >   adsErrorInfo:
04-09-2023 15:23:12.921 > error > ads_24 >    { adsErrorType: 'AMS error',
04-09-2023 15:23:12.921 > error > ads_24 >      adsErrorCode: 8,
04-09-2023 15:23:12.921 > error > ads_24 >      adsErrorStr: 'Unknown command ID' },
04-09-2023 15:23:12.921 > error > ads_24 >   metaData: null,
04-09-2023 15:23:12.921 > error > ads_24 >   errorTrace:
04-09-2023 15:23:12.921 > error > ads_24 >    [ 'readSystemManagerState(): Device system manager state read failed',
04-09-2023 15:23:12.921 > error > ads_24 >      '_sendAdsCommand(): Response with AMS error received' ],
04-09-2023 15:23:12.921 > error > ads_24 >   getInnerException: [Function] }
```

ptorrent, 2023-09-04:
For your information, I'm using an export of symbols:

```
<PlcProjectInfo>
	<RoutingInfo>
		<AdsInfo>
			<NetId>192.168.1.232.1.1</NetId>
			<Port>100</Port>
		</AdsInfo>
	</RoutingInfo>
	<Symbols>
		<Symbol>
			<Name>KL6781-0000_1_Channel_1.State</Name>
			<Type>UINT</Type>
			<IGroup>61472</IGroup>
			<IOffset>0</IOffset>
			<BitSize>16</BitSize>
		</Symbol>
...
```

I'm trying to subscribe to the symbole name and I got this error "Index out of range".

So I tried to get raw value with groupIndex and offset but same error !

The result of a read is: (buffer is the value, USINT is the datatype from export )

04-09-2023 20:15:58.997 > error > ads_24 > <Buffer 00> USINT
04-09-2023 20:15:58.997 > error > ads_24 > <Buffer 00> USINT
04-09-2023 20:15:59.009 > error > ads_24 > <Buffer 00> USINT
04-09-2023 20:15:59.009 > error > ads_24 > <Buffer 00> USINT

jisotalo, 2023-09-04:
Could you please show the Node.js code you are using with connection settings?

Quickly looking the port could be also 300 (https://download.beckhoff.com/download/document/io/bus-terminals/bk9000_bk9050_bk9100en.pdf):
[image]

ptorrent, 2023-09-05:
Thanks for your answer, this is my settings:

```
 { 
	localAmsNetId: '192.168.1.110.1.1',
	localAdsPort: 32905,
	targetAmsNetId: '192.168.1.232.1.1',
	targetAdsPort: 300,
	allowHalfOpen: true,
	bareClient: true,
	readAndCacheSymbols: false,
	routerAddress: '192.168.1.232',
	routerTcpPort: 48898 
}
```

jisotalo, 2023-09-06:
What happens when you try to read it using indexGroup and indexOffset with `readRaw`?

I'm not too familiar with BK9050 and I don't have any unit to test myself.

ptorrent, 2023-09-06:
I'm able to get the value: Buffer(00 00)

```
client.connect().then(function(err){
                             client.readRaw( target.group,target.offset,target.length).then(function(res){
					===> OK I've the res

					===> crash the process on convertion
					client.convertFromRaw(res, target.type).then(function(value){
						
					})
				}).catch(function(err){
				
				})
})
```

jisotalo, 2023-09-06:
Is the value correct? Can you get a value that isn't 0?

ptorrent, 2023-09-06:
I will check it tonight and let you know, not at my desk, thanks for your support

ptorrent, 2023-09-08:
Hello again, I was able to read value by adding a try catch:

``` ligne 6049 ads-client.js
try{
  switch (packet.ams.adsCommand) {
....
  debugD(`_parseAdsData(): ADS data parsed: %o`, ads)
} catch(e){}
```

ptorrent, 2023-09-08:
But with that solutation, cannot read value with readSymbol because of this error and readRaw for BOOL type:

```
ClientException: Reading data failed (Cannot read property 'byteLength' of undefined)
line 1442
```

If you want to check a bit, I can give you access to the linux machine.

jisotalo, 2023-09-10:
@ptorrent can you send me email. You can find the address from package.json details for example.

jisotalo, 2023-09-23:
Issues related to this are now fixed in 1.14.3

closed 2023-09-23 by jisotalo