jisotalo/ads-client #121: idea: ads-client with Blockly

runtimevic, 2023-11-08:
Hi,
idea suggestion: 
does this exist ads-client with blockly?
 I think it would be very good,
thanks.

jisotalo, 2023-11-09:
Could you elaborate more? What do you mean?

runtimevic, 2023-11-09:
Hi @jisotalo ,
with blocks custom and with node.js and with your library to be able to read and write the variables, and create programs...
https://blockly-demo.appspot.com/static/demos/code/index.html

runtimevic, 2023-12-24:
Hello @jisotalo , @Hopperpop:

- https://github.com/VisualPLC

- https://www.linkedin.com/posts/v%C3%ADctor-dur%C3%A1n-mu%C3%B1oz-2879a913_merry-christmas-i-am-developing-activity-7143974237270167553-Eq7E?utm_source=share&utm_medium=member_desktop

[image]

runtimevic, 2023-12-24:
Hi @jisotalo , @Hopperpop ,

- adsClient.Connect(301); --> is fault with ads-client...

[image]

- program C# not problem is OK
[image]

How can I send the connection without address, only with port?
thanks Víctor.

runtimevic, 2024-01-04:
Hello @jisotalo , @Hopperpop 
How can I send the connection without address, only with port 301?
thanks Víctor.

jisotalo, 2024-01-07:
Sorry @runtimevic , I don't understand what do you mean? If you leave [localAmsNetId](url) empty (or set is as localhost) it will connect to the local runtime.

runtimevic, 2024-01-07:
Hi @jisotalo 
not work localAmsNetId = localhost
port = 301
[image]
[image]
[image]

runtimevic, 2024-01-07:
Hi,
I finally solved it:
```javascript
  static async connect(address:string, port:number){
          
    Environment.client = new ads.Client({
      targetAmsNetId: address, //or 'localhost'
      targetAdsPort: port,
      bareClient: true, //NOTE
    })
````

Hopperpop, 2024-01-07:
The problem is that with default settings it checks if the PLC runtime is running. As you don't have one configured it causes errors.
For that you have the `allowHalfOpen` and `bareClient` option. As i understood `allowHalfOpen` will still checks if it finds the system manager to see if the system is in RUN or CONFIG. With bareClient is also skips that check. (@jisotalo Correct me if I'm wrong.)

runtimevic, 2024-01-07:
Hi @Hopperpop , @jisotalo 
I solved it this way by adding the bareClient parameter:

[image]

If you like the project and want to collaborate in the organization, you are welcome...

- https://github.com/VisualPLC

- https://www.linkedin.com/posts/v%C3%ADctor-dur%C3%A1n-mu%C3%B1oz-2879a913_merry-christmas-i-am-developing-activity-7143974237270167553-Eq7E?utm_source=share&utm_medium=member_desktop

jisotalo, 2024-01-07:
Good that you solved it!!

@Hopperpop you absolutely right, thanks again!

closed 2024-01-07 by jisotalo