jisotalo/ads-client #123: Get events/alarms/messages from TcEventLogger

chrissyheb, 2023-11-13:
Hi everyone,
I'm implementing a JavaScript visualization for a TwinCAT PLC, where I am using the Eventlogger to create/send alarm messages. 
Beckhoff provides only a C# library for interaction with TcEventlogger. Is the reception of event logger messages possible with ads-client? Is there another possibility to receive the event messages/alarms with JavaScript?
Thanks
Chrissy

jisotalo, 2023-11-14:
Hi

I have created a code for twincat logger (NOT event logger).
https://gist.github.com/jisotalo/80b01d590666dc19b228c732353820a1

It should be possible to create one for TcEventLogger as well. Perhaps somebody could create that, unfortunately I don't have spare time to work on it without a cost at the moment.

If you are located in Europe, I am also able to invoice, if you would like to have a paid solution for this. You can send me email if you are interested.

chrissyheb, 2023-11-15:
Hi,
thank you for the hint. I will take a look and try to implement a solution.
I will post it here if I get it working.

jisotalo, 2023-11-15:
👍 Check also this post: https://jisotalo.fi/subscribing-to-twincat-logger-in-nodejs/

closed 2024-07-11 by jisotalo