jisotalo/ads-client #125: ads-client+vite+vanillajs

runtimevic, 2024-01-21:
Hello, 
I'm trying ads-client+vite+vanillajs
package.json:
```
{
  "name": "vite-project",
  "private": true,
  "version": "0.0.0",
  "type": "module",
  "scripts": {
    "dev": "vite",
    "build": "vite build",
    "preview": "vite preview"
  },
  "devDependencies": {
    "vite": "^5.0.8"
  },
  "dependencies": {
    "ads-client": "^1.14.3"
  }
}
```
[image]

main.js:
```
import  ads  from 'ads-client';
//const ads = require('ads-client')

const client = new ads.Client({
  targetAmsNetId: '127.0.0.1.1.1',
  targetAdsPort: 851
})

client.connect()
  .then(res => {   
    console.log(`Connected to the ${res.targetAmsNetId}`)
    console.log(`Router assigned us AmsNetId ${res.localAmsNetId} and port ${res.localAdsPort}`)

    return client.disconnect()
  })
  .then(() => {
    console.log('Disconnected')
  })
  .catch(err => {
    console.log('Something failed:', err)
  })

/*
Example console output:

Connected to the 127.0.0.1.1.1
Router assigned us AmsNetId 192.168.1.1.1.1 and port 36837
Disconnected
*/
```

Is this combination vite+vanillajs possible?

jisotalo, 2024-01-21:
Hi! It's not possible as the ads-client can't run in browser.

closed 2024-01-21 by jisotalo

runtimevic, 2024-01-21:
Hi @jisotalo ,
What do you recommend all in one frontend + backend in a single app with javascript or typescript, to put ads-client, blockly, sqlite, mqtt, etc.??

benhar-dev, 2024-01-21:
Electron

runtimevic, 2024-01-22:
> Electron

Hi @benhar-dev ,
https://github.com/VisualPLC/Blockly-Electron
help me...

jisotalo, 2024-01-22:
I'm sorry but this is going a little too off-topic. The technology stack depends on so many things.

I have done few projects using ads-client that use React as frontend and then it communicates to backend using both webosckets and REST api.  And I have also used SQLite, Postgres etc.