jisotalo/ads-client #142: Question about subscriptions

ptorrent, 2024-08-20:
Hello there,

I have a very fast-moving value (input 1/0). How does registration work? Is it polling? Is it an event sent by the automaton when the value changes?

I suppose that polling at less than 1 second is not recommanded?

Thanks for your support :)

crishoj, 2024-08-20:
Subscriptions are a built-in feature of the Beckhoff ADS protocol, and follows [publish-subscribe](https://en.wikipedia.org/wiki/Publish–subscribe_pattern) semantics, not polling.

Check the [documentation](https://jisotalo.fi/ads-client/Client.html#subscribe) for `subscribe()`.

> I suppose that polling at less than 1 second is not recommanded?

ADS subscriptions can easily provide updates 10 or even 100 times per second.

jisotalo, 2024-08-21:
Thanks @crishoj!

I'm just working on documentation of v2 and I will try to improve this too.

It's indeed so that PLC checks for the value changes and sends the data when needed. There is no polling, ads-client will not send any requests at all.

I just lately used ADS notifications to monitor modbus data from 1ms cycle, using cycle time of 0. It worked just fine. However that was with .NET library but I have tested ads-client with 1ms before as well.

So in short, use as small cycle time as you feel that is needed.

closed 2024-08-30 by jisotalo