jisotalo/ads-client #143: Version 2 — publish pre-release version

crishoj, 2024-08-22:
Would be nice to be able to install V2 with `npm install ads-client@beta`.

From https://gist.github.com/adcreare/00a10f389841dd4c97197608c9b33a7f:

> Steps to publish a npm package to beta that won't be available via latest and won't auto install on ncu updates etc
> 
> 1. Ensure any compile is run `npm run dist` etc
> 2. Modify version in package.json to the following format (match with existing verion numbers etc) `"version": "0.1.120-beta.1"` where beta.x is the number of those betas
> 3. Publish to npm `npm publish --tag beta`
> 
> There are two options for install:
> * Always install beta with `npm install packagename@beta`
> * Install specific version with `npm install package@0.1.120-beta.1`
> 
> How to fix latest if you publish a beta without --tag beta (if will default to latest)
> run: `npm dist-tags add @COMPANY/PACKAGE@1.0.3 latest --otp=123456`where 1.0.3 is the version that should be latest --otp= is required

jisotalo, 2024-08-24:
Will do that! I still have some breaking changes to commit. After that I will do a beta release.

jisotalo, 2024-08-24:
@crishoj released!

https://www.npmjs.com/package/ads-client/v/2.0.0-beta.1

closed 2024-08-26 by jisotalo