jisotalo/ads-client #144: (v2) TypeError in `onAdsCommandReceived`

crishoj, 2024-08-26:
Observed after reconnecting to running PLC with V2:

```
🛜 [plc SAFETY] Initiating PLC connection to AMSNetId 10.3.200.1.1.1 IP 10.3.200.2 port 48898 ...                                                 10:47:25.608
1417 |                 for (let stamp of packet.ads.payload.stamps) {
1418 |                     //Stamp has n samples inside
1419 |                     for (let sample of stamp.samples) {
1420 |                         //Try to find the subscription
1421 |                         const key = ADS.amsAddressToString(packet.ams.sourceAmsAddress);
1422 |                         const subscription = this.activeSubscriptions[key][sample.notificationHandle];
                                                                             ^
TypeError: undefined is not an object (evaluating 'this.activeSubscriptions[key][sample.notificationHandle]')
      at onAdsCommandReceived (.../node_modules/ads-client/dist/ads-client.js:1422:71)
      at onAmsTcpPacketReceived (.../node_modules/ads-client/dist/ads-client.js:1348:26)
      at parseAmsTcpPacket (.../node_modules/ads-client/dist/ads-client.js:1104:14)
```

jisotalo, 2024-08-26:
Hopefully now fixed in beta.2!

crishoj, 2024-08-30:
> I think this issue is possible when you first subscribe to a variable and then the connection closes without unsubscribing (so the PLC keeps sending notifications).

Makes sense. A situation with stale subscription handles could arise if e.g. the application crashes during development, before `unsubscribe()` was called.

> Hopefully now fixed in beta.2!

Thanks!

jisotalo, 2024-08-30:
Closing this for now. Let me know if there are still any issues :)

closed 2024-08-30 by jisotalo