jisotalo/ads-client #155: convertFromRaw does not correct check the size for arrays of structs

InconceivableDuck, 2024-09-17:
Related to https://github.com/jisotalo/ads-client/issues/53

Looks like it was fixed for arrays of basic types, but not for arrays of complex types.

Error:

```
ClientException: Given data buffer and data type sizes mismatch: Buffer is 16192 bytes and data type is 2484 bytes
```

Symbol Info:

```
{
  indexGroup: 16448,
  indexOffset: 841576,
  size: 16192,
  adsDataType: 65,
  adsDataTypeStr: 'ADST_BIGTYPE',
  flags: 8,
  flagsStr: [ 'TypeGuid' ],
  arrayDimension: 0,
  nameLength: 25,
  typeLength: 39,
  commentLength: 0,
  name: 'GVL_Var.TestArrayStruct40',
  type: 'ARRAY [0..10,15..30] OF DUT_Structure_1',
  comment: '',
  arrayData: [],
  typeGuid: 'efa3486981b90e95fbbf485c484d6131',
  attributes: [],
  reserved: <Buffer 00 00 00 00 00 00 00>
}
```

Result from `readRawMulti`:

```
{
  success: true,
  errorInfo: { error: false, errorCode: 0, errorStr: 'No error' },
  target: { indexGroup: 16448, indexOffset: 841576 },
  data: <Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ... 16142 more bytes>
}
```

We can see the raw data's length matches the symbol's size, but the check errors.

crishoj, 2024-09-18:
I also encountered an issue with arrays of structs. I believe it is fixed in the upcoming v2.0, in case you'd like to give that a try:

```
 "ads-client": "^2.0.0-beta.3"
```

Migration guide: https://github.com/jisotalo/ads-client/blob/2.0.0-beta.3/MIGRATION.md

InconceivableDuck, 2024-09-19:
I can confirm that v2 does work.

jisotalo, 2024-09-21:
Thanks for reporting! If everything works in v2, then I might not look into this.

InconceivableDuck, 2024-09-26:
Happy to wait for the v2 release. Closing this issue.

closed 2024-09-26 by InconceivableDuck