jisotalo/ads-client #157: (v2-dev) `disconnect()` hangs (doesn't resolve) when using connection Setup 2

crishoj, 2024-09-28:
I observed a surprising behaviour using V2 and [connection Setup 2](https://github.com/jisotalo/ads-client?tab=readme-ov-file#setup-2---connecting-from-unixwindowsetc-system-to-the-plc) (with a local `AdsRouterConsole`:

- `connect()` works as expected
- but the promise returned from `disconnect()` appears to never resolve

Debug log:

```
  ads-client disconnectFromTarget(): Disconnecting (force: false) +668ms
  ads-client removeSubscriptions(): Unsubscribing from all active subscriptions (also internals: true) +0ms
  ads-client unsubscribe(): Unsubscribing { settings: { target: { indexGroup: 61696, indexOffset: 0, size: 4 }, callback: [Function: bound onPlcRuntimeStateChanged], cycleTime: 0 }, internal: true, remoteAddress: { amsNetId: '10.3.0.20.1.1', adsPort: 852 }, notificationHandle: 15, symbol: undefined, unsubscribe: [AsyncFunction: unsubscribe], parseNotification: [AsyncFunction: parseNotification], latestData: { timestamp: 2024-09-28T13:51:11.824Z, value: Buffer(4) [Uint8Array] [ 5, 0, 0, 0 ] }, targetOpts: {} } +0ms
  ads-client deleteNotificationHandle(): Sending DeleteNotification command for handle 15 to 10.3.0.20.1.1:852 +1ms
  ads-client deleteNotificationHandle(): Notification for handle 15 (10.3.0.20.1.1:852) deleted! +14ms
  ads-client unsubscribe(): Subscription with handle 15 unsubscribed from 10.3.0.20.1.1:852 +0ms
  ads-client unsubscribe(): Unsubscribing { settings: { target: { indexGroup: 61448, indexOffset: 0, size: 1 }, callback: [Function: bound onPlcSymbolVersionChanged] AsyncFunction, cycleTime: 0 }, internal: true, remoteAddress: { amsNetId: '10.3.0.20.1.1', adsPort: 852 }, notificationHandle: 16, symbol: undefined, unsubscribe: [AsyncFunction: unsubscribe], parseNotification: [AsyncFunction: parseNotification], latestData: { timestamp: 2024-09-28T13:51:11.864Z, value: Buffer(1) [Uint8Array] [ 11 ] }, targetOpts: {} } +0ms
  ads-client deleteNotificationHandle(): Sending DeleteNotification command for handle 16 to 10.3.0.20.1.1:852 +0ms
  ads-client deleteNotificationHandle(): Notification for handle 16 (10.3.0.20.1.1:852) deleted! +8ms
  ads-client unsubscribe(): Subscription with handle 16 unsubscribed from 10.3.0.20.1.1:852 +3ms
  ads-client removeSubscriptions(): Unsubscribed from all 2 subscriptions successfully +0ms
  ads-client disconnectFromTarget(): Disconnecting (force: false) +11s
  ads-client removeSubscriptions(): Unsubscribing from all active subscriptions (also internals: true) +0ms
  ads-client removeSubscriptions(): Unsubscribed from all 0 subscriptions successfully +0ms
```

With [connection Setup 3](https://github.com/jisotalo/ads-client?tab=readme-ov-file#setup-3---connecting-from-any-nodejs-supported-system-to-the-plc) (direct connection from Mac to TwinCAT on Windows), `disconnect()` work as expected, and resolves after the connection is closed.

Debug log with connection Setup 3:

```
  ads-client disconnectFromTarget(): Disconnecting (force: false) +532ms
  ads-client removeSubscriptions(): Unsubscribing from all active subscriptions (also internals: true) +0ms
  ads-client unsubscribe(): Unsubscribing { settings: { target: { indexGroup: 61696, indexOffset: 0, size: 4 }, callback: [Function: bound onPlcRuntimeStateChanged], cycleTime: 0 }, internal: true, remoteAddress: { amsNetId: '10.3.0.20.1.1', adsPort: 851 }, notificationHandle: 202, symbol: undefined, unsubscribe: [AsyncFunction: unsubscribe], parseNotification: [AsyncFunction: parseNotification], latestData: { timestamp: 2024-09-28T13:51:00.594Z, value: Buffer(4) [Uint8Array] [ 5, 0, 0, 0 ] }, targetOpts: {} } +0ms
  ads-client deleteNotificationHandle(): Sending DeleteNotification command for handle 202 to 10.3.0.20.1.1:851 +1ms
  ads-client deleteNotificationHandle(): Notification for handle 202 (10.3.0.20.1.1:851) deleted! +16ms
  ads-client unsubscribe(): Subscription with handle 202 unsubscribed from 10.3.0.20.1.1:851 +0ms
  ads-client unsubscribe(): Unsubscribing { settings: { target: { indexGroup: 61448, indexOffset: 0, size: 1 }, callback: [Function: bound onPlcSymbolVersionChanged] AsyncFunction, cycleTime: 0 }, internal: true, remoteAddress: { amsNetId: '10.3.0.20.1.1', adsPort: 851 }, notificationHandle: 203, symbol: undefined, unsubscribe: [AsyncFunction: unsubscribe], parseNotification: [AsyncFunction: parseNotification], latestData: { timestamp: 2024-09-28T13:51:00.624Z, value: Buffer(1) [Uint8Array] [ 33 ] }, targetOpts: {} } +0ms
  ads-client deleteNotificationHandle(): Sending DeleteNotification command for handle 203 to 10.3.0.20.1.1:851 +0ms
  ads-client deleteNotificationHandle(): Notification for handle 203 (10.3.0.20.1.1:851) deleted! +10ms
  ads-client unsubscribe(): Subscription with handle 203 unsubscribed from 10.3.0.20.1.1:851 +1ms
  ads-client removeSubscriptions(): Unsubscribed from all 2 subscriptions successfully +0ms
  ads-client unregisterAdsPort(): Local AmsNetId and ADS port manually provided, no need to unregister +0ms
  ads-client disconnectFromTarget(): Connection closed successfully +0ms
```

crishoj, 2024-09-28:
I assume it's `unregisterAdsPort()` that fails to resolve.

Could it be that the the following assumption doesn't hold when connected through `AdsRouterConsole`?

> The remote seems to drop TCP connection after unregistering, 
> so basically calling this always disconnects too.

jisotalo, 2024-09-29:
Hi @crishoj!

That's odd. I need to test the AdsRouterConsole, haven't used it for a while.

By looking at the source code, I can't see a command to close the TCP connection. Instead, the if I undestand correctly, the router would perhaps send a response (or is it just in error situtation?):

```c#
 private async Task<AdsErrorCode> onPortCloseRequest(
      TcpPortCloseRequestFrame frame,
      TcpClientConnectionInfo info,
      CancellationToken cancel)
    {
      AmsTcpIpRouter amsTcpIpRouter1 = this;
      AmsAddress key = new AmsAddress(AmsNetId.Local, (int) frame.Port);
      TcpClientConnectionInfo info2;
      bool flag = amsTcpIpRouter1._loopbackConnections.TryGetValue(key, out info2);
      AmsTcpIpRouter amsTcpIpRouter2 = amsTcpIpRouter1;
      DefaultInterpolatedStringHandler interpolatedStringHandler1 = new DefaultInterpolatedStringHandler(56, 2);
      interpolatedStringHandler1.AppendLiteral("Requesting for PortClose of Address '");
      interpolatedStringHandler1.AppendFormatted<AmsAddress>(key);
      interpolatedStringHandler1.AppendLiteral("' (LoopbackFound: ");
      interpolatedStringHandler1.AppendFormatted<bool>(flag);
      interpolatedStringHandler1.AppendLiteral(")");
      string stringAndClear1 = interpolatedStringHandler1.ToStringAndClear();
      object[] objArray1 = Array.Empty<object>();
      LogProviderExtension.LogInformation((ILoggerProvider) amsTcpIpRouter2, stringAndClear1, objArray1);
      AdsErrorCode errorCode = amsTcpIpRouter1.OnUnregisterServer(frame.Port);
      AdsErrorCode adsErrorCode1;
      if (flag)
      {
        adsErrorCode1 = await amsTcpIpRouter1.onPortCloseResponse(errorCode, info2, cancel);
        info2.Cancel();
      }
      else
        adsErrorCode1 = (AdsErrorCode) 1;
      if (AdsErrorCodeExtension.Failed(adsErrorCode1))
      {
        AmsTcpIpRouter amsTcpIpRouter3 = amsTcpIpRouter1;
        DefaultInterpolatedStringHandler interpolatedStringHandler2 = new DefaultInterpolatedStringHandler(34, 1);
        interpolatedStringHandler2.AppendLiteral("Port close response failed with '");
        interpolatedStringHandler2.AppendFormatted<AdsErrorCode>(adsErrorCode1);
        interpolatedStringHandler2.AppendLiteral("'");
        string stringAndClear2 = interpolatedStringHandler2.ToStringAndClear();
        object[] objArray2 = Array.Empty<object>();
        LogProviderExtension.LogError((ILoggerProvider) amsTcpIpRouter3, stringAndClear2, objArray2);
      }
      AdsErrorCode adsErrorCode2 = adsErrorCode1;
      info2 = (TcpClientConnectionInfo) null;
      return adsErrorCode2;
    }
```
Source: TwinCAT.Ads.TcpRouter / `AmsTcpIpRouter.cs`. Decompiled with dotpeek.

Anyways, your PR could solve the problem, we just need to make sure it's ok to do this "fire and forget" port unregistering. 

Also, in this PR, if the server doesn't end the TCP connection, it will stay open. So we would need to manually call `socket.end()` and/or `socket.destroy()` before resolving.

Perhaps if the socket is not ended by remote in x seconds, we could fall back to manually ending it and then resolving?

crishoj, 2024-09-29:
> I can't see a command to close the TCP connection. Instead, the if I undestand correctly, the router would perhaps send a response (or is it just in error situtation?)

Let me try again with debug level 3, assuming this will reveal the response from `AdsRouteConsole`, if any.

jisotalo, 2024-09-29:
Perhaps it's this one?
https://github.com/jisotalo/ads-client/blob/9ea67f944a631b4175d62177ed73a8df869fd327/src/ads-client.ts#L1740C32-L1740C50

crishoj, 2024-09-29:
Could `AMS_TCP_PORT_CLOSE` be the packet that `unregisterAdsPort` should react to?

```
  ads-client:details unregisterAdsPort(): Unregistering ADS port 32772 from target :48898 +1ms
  ads-client:raw-data IO out ------> 8 bytes : 0100020000000480 +1ms
  ads-client:raw-data IO in  <------ 10 bytes from 127.0.0.1: 01000400000000000000 +3ms
  ads-client:details parseAmsTcpHeader(): Starting to parse AMS/TCP header +3ms
  ads-client:details parseAmsTcpHeader(): AMS/TCP header parsed: { command: 1, commandStr: 'AMS_TCP_PORT_CLOSE', length: 4, data: Buffer(4) [Uint8Array] [ 0, 0, 0, 0 ] } +0ms
  ads-client:details parseAmsHeader(): Starting to parse AMS header +0ms
  ads-client:details parseAmsHeader(): No AMS header found +0ms
  ads-client:details parseAdsResponse(): Starting to parse ADS data +0ms
  ads-client:details parseAdsData(): Packet has no ADS data +0ms
  ads-client:details onAmsTcpPacketReceived(): AMS packet received with command 1 +0ms
```

crishoj, 2024-09-29:
> Perhaps it's this one?

Oh yes, indeed — just saw your comment.

crishoj, 2024-09-29:
#158 updated to handle incoming `AMS_TCP_PORT_CLOSE` packets.

jisotalo, 2024-10-02:
Closing this as completed

closed 2024-10-02 by jisotalo