jisotalo/ads-client #159: `restartPlc` method can cause stale `Stop` state in `metaData.plcRuntimeState`

crishoj, 2024-10-29:
Not a huge problem by any means, but thought I would mention the observation:

`restartPlc()` does a _cold reset_ followed by a _start_ in rapid succession, and seems to sometimes miss the fact that the PLC returns to `Run` state:

```ts
      await this.resetPlc(targetOpts);
      await this.startPlc(targetOpts);

      this.debug(`restartPlc(): PLC runtime restarted at ${this.targetToString(targetOpts)}`);
```

In the project where I use `ads-client`, I ended up doing the following instead, which seems to work reliably:

1. _cold reset_ using `resetPlc`
2. wait for runtime state be `Stop`
3. _start_ using `startPlc`
4. wait for runtime state be `Run` 

Using a `waitFor` primitive that checks the passed predicate whenever `plcRuntimeStateChange` fires, it looks like this:

```ts
    await this.ads.resetPlc()
    await this.waitFor({ predicate: () => this.runtimeState === 'Stop' })
    await this.ads.startPlc()
    await this.waitFor({ predicate: () => this.runtimeState === 'Run', timeoutMs: 3000 })
```

jisotalo, 2024-10-31:
Thanks!

Strange that during testing everything worked fine, even with real hardware. When looking at the test code now, I can see it reads the state manually, instead of checking `this.ads.metaData.plcRuntimeState?.adsState`. Perhaps the value in metaData isn't updating in all cases.

Need to check this out.

```js
test('restarting PLC', async () => {
    let state = await client.readPlcRuntimeState();
    expect(state.adsState).toBe(ADS.ADS_STATE.Run);
    await client.writeValue('GVL_AdsClientTests.IsReset', false);

    await client.restartPlc();

    //Some delay as PLC was just started
    await delay(500);

    state = await client.readPlcRuntimeState();
    expect(state.adsState).toBe(ADS.ADS_STATE.Run);

    const res = await client.readValue('GVL_AdsClientTests.IsReset');
    expect(res.value).toBe(true);
  });
```

jisotalo, 2024-11-03:
@crishoj I'm using a wrong issue to ask, but do you happen to have a TC3 4026 available?

I think it's the only thing to test before releasing v2 stable (when README is ready). I have no access to 4026 PLC for now, but I will try to install the XAR to my other laptop. I might try to install 4026 to one PLC some day, but just asking if you would have it already :)

crishoj, 2024-11-03:
> Perhaps the value in metaData isn't updating in all cases.

This is also my impression.

jisotalo, 2024-11-24:
I improved the client to handle these cases better. Could you test? 

I update test cases to check `metaData.plcRuntimeState.adsState` also. After that, I got some failures. But after these fixes the tests pass nicely.

Basically now `startPlc()`, `resetPlc()`, `stopPlc()` and `restartPlc()` read active runtime state afterwards to update metadata. Also `readPlcRuntimeState()` now emits the `plcRuntimeStateChange` event if the state has changed. Previously it just updated the metadata, which could cause issues like this if the notification wasn't yet received.

**From changelog:**
- When calling `readPlcRuntimeState()` and state has changed, event `plcRuntimeStateChange` is emitted (previously emitted only if notification received from PLC)
  - See [issue #159](https://github.com/jisotalo/ads-client/issues/159)
- `startPlc()`, `resetPlc()`, `stopPlc()` and `restartPlc()` call `readPlcRuntimeState()` afterwards to update the runtime state to metadata (and to emit event if needed) 
  - See [issue #159](https://github.com/jisotalo/ads-client/issues/159)

crishoj, 2024-11-25:
Thank you @jisotalo. May I ask you to publish a new v2 beta?

jisotalo, 2024-11-25:
Will do, I'll see if I could fix #163 before that

jisotalo, 2024-11-25:
@crishoj 2.0.0-beta.5 released!

crishoj, 2024-11-26:
Thank you @jisotalo!

Our tests for restarting (_reset cold_) and stop/starting the PLC repeatedly are passing now:

[image]

jisotalo, 2024-11-26:
Thats's great!

I think v2 stable release is not that far away anymore. Just some documentation stuff.