jisotalo/ads-client #161: Adding support for upload info flags and UTF-8 encoded ADS symbols

jisotalo, 2024-11-17:
This is related to https://github.com/jisotalo/ads-client/issues/147

Symbol-related tests are failing when connecting to TwinCAT 4026.
[image]

The reason is that TwinCAT 4026 has UTF-8 encoded ADS values as default. In older versions, that can be achieved from settings under SYSTEM.

[image]

- https://infosys.beckhoff.com/english.php?content=../content/1033/tf6730_tc3_iot_communicator/17258234891.html&id=5886726884932349632
- https://infosys.beckhoff.com/english.php?content=../content/1033/tf6701_tc3_iot_communication_mqtt/4220231435.html&id=

When digging into TwinCAT.Ads.dll, we can see that is no way to know if the specific string is encoded in utf8 or not:
```c#
  /// Since TwinCAT 4026, the UTF8 Encoding will be the default
  /// Historically in older TwinCAT installations, the Default encoding is ANSI with Windows active (e.g. CodePage 1252 (West-European))
  /// Because the AdsClient doesn't know what the actual communication partner is (4026 or older) we try to default to ANSI.
  /// The AdsClient holds its own actual encoding in SymbolNameEncoding/SymbolNameMarshaler with the (ADS) read settings from SymbolUploadInfo.
```
_Source: TwinCAT.Ads.Abstractions.StringMarshaler_

So this means that the client needs to read the target upload info version and if it's >= 3, also read if utf8 is used. 

Current `readPlcUploadInfo()` solution:
https://github.com/jisotalo/ads-client/blob/6aa28e15beea0a2df0567b796b02f2c916d72f9f/src/ads-client.ts#L4870

We need to read upload info flags to check for `Utf8EncodedStringData` flag. However we also need to make sure support for TC2 and TC3 <= 4024 work as before.

```c#
public enum SymbolUploadFlags : uint
{
  /// <summary>None / Unititialized</summary>
  None = 0,
  /// <summary>Target is 64 Bit Platform</summary>
  Is64BitPlatform = 1,
  /// <summary>Symbol Server includes Base types.</summary>
  IncludesBaseTypes = 2,
  /// <summary>String Date is encoded in UTF8</summary>
  /// <remarks>
  /// All PLC STRING data is encoded in UTF8. This flag is used from TwinCAT 4026 on.
  /// </remarks>
  Utf8EncodedStringData = 4,
}
```
_Source: TwinCAT.Ads.Internal_

jisotalo, 2024-11-17:
Adding also support for older upload info versions. At the moment, the client is fixed to v2.

The C# library handles this by reading from index group 61455 (v2 and v3) and if it fails, then from 61452 (v1).

crishoj, 2024-11-17:
Then `decodePlcStringBuffer` and `encodeStringToPlcStringBuffer` need to take the flag `Utf8EncodedStringData` into account.

With this change, I imagine some `iconv`-related overhead on the client side can be eliminated by using UTF-8 on the PLC.

jisotalo, 2024-11-24:
Support added in this [commit](https://github.com/jisotalo/ads-client/commit/9a90734ef1786260f93f7039f27607473500e72d).

Previously failed tests pass now in 4026 and also in 4024 with UTF-8 enabled on ADS symbols.

Note: This doesn't support UTF-8 encoded strings yet, that's another issue (https://github.com/jisotalo/ads-client/issues/162).

closed 2024-11-24 by jisotalo