jisotalo/ads-client #165: Tests for handling unknown/stale ADS notifications failing

crishoj, 2024-11-30:
Running `test/TC3/ads-client.test.js` against my PC-based TC3 4024.62, all tests are passing, except the tests related to handling unknown/stale ADS notifications.

The problem appears to be connecting the auxiliary client:

```
connectToTarget(): Connection failed - failed to set PLC connection. If target is not PLC runtime, use setting "rawClient". If system is in config mode or there is no PLC software yet, you might want to use setting "allowHalfOpen"
ClientError: readPlcUploadInfo(): Reading upload info failed
ClientError: sendAdsCommand(): Response with ADS error received (error 1808 - Symbol not found)
```

I wonder what prevents the second client from connecting. I am using connection [setup 2](https://github.com/jisotalo/ads-client?tab=readme-ov-file#setup-2---connecting-from-unixwindowsetc-system-to-the-plc), with ads-client + `AdsRouterConsole` on my Mac, connecting to TC on a Windows PC.

Any idea what's preventing the second client from connecting?

jisotalo, 2024-12-02:
Hmm, this is odd. Are the tests under `controlling TwinCAT system service` passing? So is the only issue the client under stale notification handle tests.

https://github.com/jisotalo/ads-client/blob/51702a2a494d32b3678afbfb001fb6fd0f2236dd/test/TC3/ads-client.test.js#L3595

Could you also set the debugging level to 2 and paste the logs.

jisotalo, 2024-12-09:
@crishoj I just got the same issue and the reason was that **Autostart boot project** was not checked.

Can you check that?

crishoj, 2024-12-09:
Just made sure _Autostart boot project_ is checked:

[image]

Still getting two failing tests:

[image]

And the test hangs at the end with this message (from `jest` option `--detectOpenHandles`):

[image]

crishoj, 2024-12-09:
> Could you also set the debugging level to 2 and paste the logs.

<details>
<summary>Debug output</summary>

```
~/C/ads-client (v2-dev|✚1) $ DEBUG='ads-client,ads-client:details' AMS_NET_ID=100.70.46.31.1.1 LOCAL_AMS_NET_ID=100.117.55.75.1.1 jest --runInBand TC3/ads-client.test.js
  ads-client connectToTarget(): Connecting to target at 127.0.0.1:48898 (reconnecting: false) +0ms
  ads-client connectToTarget(): Socket connection established to 127.0.0.1:48898 +0ms
  ads-client registerAdsPort(): Registering an ADS port from target router (127.0.0.1:48898) +1ms
  ads-client:details IO out ------> 8 bytes +0ms
  ads-client:details IO in  <------ 14 bytes from 127.0.0.1 +0ms
  ads-client:details parseAmsTcpHeader(): Starting to parse AMS/TCP header +0ms
  ads-client:details parseAmsTcpHeader(): AMS/TCP header parsed: { command: 4096, commandStr: 'AMS_TCP_PORT_CONNECT', length: 8, data: <Buffer 64 75 37 4b 01 01 1a 80> } +0ms
  ads-client:details parseAmsHeader(): Starting to parse AMS header +0ms
  ads-client:details parseAmsHeader(): No AMS header found +0ms
  ads-client:details parseAdsResponse(): Starting to parse ADS data +1ms
  ads-client:details parseAdsData(): Packet has no ADS data +0ms
  ads-client:details onAmsTcpPacketReceived(): AMS packet received with command 4096 +0ms
  ads-client registerAdsPort(): ADS port registered, assigned AMS address is 100.117.55.75.1.1:32794 +1ms
  ads-client connectToTarget(): ADS port registered. Our AMS address is 100.117.55.75.1.1:32794 +0ms
  ads-client readTcSystemState(): Reading TwinCAT system state +0ms
  ads-client:details sendAdsCommand(): Sending an ads command ReadState (0 bytes): { amsTcp: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD' }, ams: { targetAmsAddress: { amsNetId: '100.70.46.31.1.1', adsPort: 10000 }, sourceAmsAddress: { amsNetId: '100.117.55.75.1.1', adsPort: 32794 }, adsCommand: 4, adsCommandStr: 'ReadState', stateFlags: 4, stateFlagsStr: 'AdsCommand, Tcp, Request', dataLength: 0, errorCode: 0, invokeId: 0 }, ads: { payload: undefined } } +0ms
  ads-client:details createAmsHeader(): AMS header created (32 bytes) +0ms
  ads-client:details createAmsTcpHeader(): AMS/TCP header created (6 bytes) +0ms
  ads-client:details createAmsTcpRequest(): AMS/TCP request created (38 bytes) +0ms
  ads-client:details IO out ------> 38 bytes +0ms
  ads-client:details IO in  <------ 46 bytes from 127.0.0.1 +18ms
  ads-client:details parseAmsTcpHeader(): Starting to parse AMS/TCP header +0ms
  ads-client:details parseAmsTcpHeader(): AMS/TCP header parsed: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD', length: 40 } +0ms
  ads-client:details parseAmsHeader(): Starting to parse AMS header +0ms
  ads-client:details parseAmsHeader(): AMS header parsed: { targetAmsAddress: { amsNetId: '100.117.55.75.1.1', adsPort: 32794 }, sourceAmsAddress: { amsNetId: '100.70.46.31.1.1', adsPort: 10000 }, adsCommand: 4, adsCommandStr: 'ReadState', stateFlags: 5, stateFlagsStr: 'Response, AdsCommand, Tcp', dataLength: 8, errorCode: 0, invokeId: 0, error: false, errorStr: '' } +0ms
  ads-client:details parseAdsResponse(): Starting to parse ADS data +0ms
  ads-client:details parseAdsResponse(): ADS data parsed: { errorCode: 0, payload: { adsState: 5, adsStateStr: 'Run', deviceState: 1 } } +0ms
  ads-client:details onAmsTcpPacketReceived(): AMS packet received with command 0 +0ms
  ads-client:details onAdsCommandReceived(): ADS command received (command: 4) +0ms
  ads-client:details sendAdsCommand(): Response received for command "ReadState" with invokeId 0 +1ms
  ads-client readTcSystemState(): TwinCAT system state read successfully. State is { adsState: 5, adsStateStr: 'Run', deviceState: 1 } +19ms
  ads-client:details addSubscription(): Subscribing { target: { indexGroup: 61696, indexOffset: 0, size: 4 }, callback: [Function: bound onPlcRuntimeStateChanged], cycleTime: 0 } (internal: true) +0ms
  ads-client:details sendAdsCommand(): Sending an ads command AddNotification (40 bytes): { amsTcp: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD' }, ams: { targetAmsAddress: { amsNetId: '100.70.46.31.1.1', adsPort: 851 }, sourceAmsAddress: { amsNetId: '100.117.55.75.1.1', adsPort: 32794 }, adsCommand: 6, adsCommandStr: 'AddNotification', stateFlags: 4, stateFlagsStr: 'AdsCommand, Tcp, Request', dataLength: 40, errorCode: 0, invokeId: 1 }, ads: { payload: <Buffer 00 f1 00 00 00 00 00 00 04 00 00 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00> } } +0ms
  ads-client:details createAmsHeader(): AMS header created (32 bytes) +0ms
  ads-client:details createAmsTcpHeader(): AMS/TCP header created (6 bytes) +0ms
  ads-client:details createAmsTcpRequest(): AMS/TCP request created (78 bytes) +0ms
  ads-client:details IO out ------> 78 bytes +0ms
  ads-client:details IO in  <------ 46 bytes from 127.0.0.1 +16ms
  ads-client:details parseAmsTcpHeader(): Starting to parse AMS/TCP header +0ms
  ads-client:details parseAmsTcpHeader(): AMS/TCP header parsed: { command: 0, commandStr: 'AMS_TCP_PORT_AMS_CMD', length: 40 } +0ms
  ads-client:details parseAmsHeader(): Starting to parse AMS header +0ms
  ads-client:details parseAmsHeader(): AMS header parsed: { targetAmsAddress: { amsNetId: '100.117.55.75.1.1', adsPort: 32794 }, sourceAmsAddress: { amsNetId: '100.70.46.31.1.1', adsPort: 851 }, adsCommand: 6, adsCommandStr: 'AddNotification', stateFlags: 5, stateFlagsStr: 'Response, AdsCommand, Tcp', dataLength: 8, errorCode: 0, invokeId: 1, error: false, errorSt [... cut ...]
      at Client.addSubscription (src/ads-client.ts:2152:13)
      at Object.<anonymous> (test/TC3/ads-client.test.js:227:20)
      ClientError: sendAdsCommand(): Timeout for command AddNotification with invokeId 10 - No response in 2000 ms
      at Timeout._onTimeout (src/ads-client.ts:3831:25)

Test Suites: 1 failed, 1 total
Tests:       2 failed, 9 passed, 11 total
Snapshots:   0 total
Time:        5.656 s, estimated 6 s
Ran all test suites matching /TC3\/ads-client.test.js/i.
```

</details>

jisotalo, 2024-12-11:
I configured AdsRouterConsole and managed to reproduce the same error. So it's definetely related to the router implementation. I'm used PC<->PC connection instead of Mac.

Delay isn't helping either.

If changing the ADS port in between, the connection works. However the tests obviously not, but this tells us that the router doesn't like using the same port immediately again

By looking at the router debug data, we can see the port is still reserved. We probably need to double check if ads-client could handle that `PortAlreadyInUse`.
```
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Port '32758' is already in use. Returning Error Code 'PortAlreadyInUse'
```

To be continued..

jisotalo, 2024-12-16:
This might be related to:
* #167 
* https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/issues/62

jisotalo, 2024-12-18:
I tested with the updated `Beckhoff.TwinCAT.Ads.TcpRouter` and the issue persists.

Maybe we are just doing things that shouldn't be done. However that test is quite good.

I'm open to any ideas