jisotalo/ads-client #167: With AdsRouterConsole, connection fails on first attempt

crishoj, 2024-12-11:
Another peculiar thing I observed when using AdsRouterConsole (development setup): 
- My app is connecting to 4 different PLC projects on the same TwinCAT PC (on different ADS ports).
- When starting up, connections to 3 of the 4 instances fail.
- If I stop my app and restart it (or attempt connection again within the app), all 4 connections succeed (and everything works).

For the failing 3 connections, `ads-client` reports:

```
Could not connect: ClientError: connectToTarget(): Connection failed - failed to set PLC connection. If target is not PLC runtime, use setting "rawClient". If system is in config mode or there is no PLC software yet, you might want to use setting "allowHalfOpen". Error: readTcSystemState(): Reading TwinCAT system state failed
```

Meanwhile, `AdsRouterConsole` reports:

```
fail: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:InternalError, Connection to target address '100.70.46.31.1.1:10000' Failed 100.117.55.75.1.1:32771 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0 (21.48.39)
fail: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:InternalError, Connection to target address '100.70.46.31.1.1:10000' Failed 100.117.55.75.1.1:32768 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0 (21.48.39)
fail: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:InternalError, Connection to target address '100.70.46.31.1.1:10000' Failed 100.117.55.75.1.1:32770 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0 (21.48.39)
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:TargetMachineNotFound, TcpClient Unknown 100.117.55.75.1.1:32771 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:TargetMachineNotFound, TcpClient Unknown 100.117.55.75.1.1:32770 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0
warn: TwinCAT.Ads.AdsRouterService.RouterService[0]
      Error:TargetMachineNotFound, TcpClient Unknown 100.117.55.75.1.1:32768 --> 100.70.46.31.1.1:10000, Cmd:ReadState REQ, State:MaskAdsRequest, Err:0, User:0, Bytes:0
```

This only occurs in my development environment, due to the use of `AdsRouteConsole`. Perhaps related to #165, but posting as a separate issue in case it is off-topic.

Initial PLC connection view within my app:

[image]

And after disconnecting/reconnecting, everything works:

[image]

I don't mean to encourage you to spend precious time looking into the issue. Just posting the details in case it could be useful.

jisotalo, 2024-12-14:
I tested with AdsRouterConsole and I'm getting similar issue

```js
const { Client } = require("../ads-client");

const client = new Client({
  targetAmsNetId: "192.168.68.106.1.1",
  targetAdsPort: 851
});

const client2 = new Client({
  targetAmsNetId: "192.168.68.106.1.1",
  targetAdsPort: 852
});

const client3 = new Client({
  targetAmsNetId: "192.168.68.106.1.1",
  targetAdsPort: 853
});

const client4 = new Client({
  targetAmsNetId: "192.168.68.106.1.1",
  targetAdsPort: 854
});

Promise.all([client.connect(), client2.connect(), client3.connect(), client4.connect()])
  .then(res => {
    console.log("Connected:", res);
  })
  .catch(err => {
    console.error("Error:", err);
  })
  .finally(() => {
    client.disconnect();
  })
```

If I start the router and run the script, i get an error. After that everything works. If I restart the router, the same issue happens again.

Could this be a AdsRouterConsole bug?

I tried to reproduce it with .NET library and got even more issues.  The .NET code starts working after restarting the router and running the Node.js code twice...

```c#

using TwinCAT.Ads;

AdsClient client = new AdsClient();
AdsClient client2 = new AdsClient();
AdsClient client3 = new AdsClient();
AdsClient client4 = new AdsClient();

client.Connect("192.168.68.106.1.1", 851);
client2.Connect("192.168.68.106.1.1", 852);
client3.Connect("192.168.68.106.1.1", 853);
client4.Connect("192.168.68.106.1.1", 854);

var tasks = new List<Task<ResultReadDeviceState>>{
        client.ReadStateAsync(default),
        client2.ReadStateAsync(default),
        client3.ReadStateAsync(default),
        client4.ReadStateAsync(default)
      };

await Task.WhenAll(tasks);

for (int i = 0; i < tasks.Count; i++)
{
  Console.WriteLine($"Task {i} - Success: {tasks[i].Result.Succeeded} - Result: {tasks[i].Result.State.AdsState.ToString()} - Error: {tasks[i].Result.ErrorCode}");
}

client.Disconnect();
client2.Disconnect();
client3.Disconnect();
client4.Disconnect();
```

crishoj, 2024-12-14:
I got similar results with the .NET library.

First run:

```
Task 1 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 2 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 3 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 4 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
```

Second run:

```
Task 1 - Success: True - Result: Run - Error: Succeeded
Task 2 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 3 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
Task 4 - Success: False - Result: Invalid - Error: ClientSyncTimeOut
```

---

If I run the .NET client with a _single_ connection once, it connects:

```
Task 1 - Success: True - Result: Run - Error: Succeeded
```

After this initial run, I can run the .NET client with 4 connections: 

```
Task 1 - Success: True - Result: Run - Error: Succeeded
Task 2 - Success: True - Result: Run - Error: Succeeded
Task 3 - Success: True - Result: Run - Error: Succeeded
Task 4 - Success: True - Result: Run - Error: Succeeded
```

I will try reporting the issue at https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples. They have been [responsive in the past](https://github.com/Beckhoff/TF6000_ADS_DOTNET_V5_Samples/issues/58).

crishoj, 2024-12-14:
Thanks for looking into this. I think it's safe to say that it's not an issue in `ads-client`.

closed 2024-12-14 by crishoj

jisotalo, 2024-12-14:
Hopefully this gets sorted out! I have very positive experience of Beckhoff .NET team, especially @RalfHeitmann

And if it turns out ads-client has something to fix, let me know.

I also tested this with 1.4.4 and results were the same, so I continued to work on v2 release and I just released it! 

Finally. 

Thanks for all the help @crishoj. I think this is a huge step and now we can just focus on improving little things.

crishoj, 2024-12-14:
Thanks for making this excellent library! It has enabled us to rapidly implement a high-level scheduling system that coordinates a number of PLC-controlled devices using Typescript.